AlexShkarin/pyLabLib #3: Arcus Stepper Motors - "single axis stage" for stepper motors

otamichalekSTFC, 2021-11-08:
Hi, thank you very much for creating this package!

I was wondering if you are planning to add functionality to connect Arcus stepper motors since the current code only seems to include 2-axis and 4-axis. I was not able to connect DMX-J-SA-17 using GenericPerformaxStage, Performax4EXStage nor Performax2EXStage. Maybe I am using it wrong? I use 64 bit dll from their website and the following:

`import pylablib as pll

pll.par["devices/dlls/arcus_performax"] = "C:/simplifiedPath/PerformaxCom.dll"

from pylablib.devices import Arcus

print(Arcus.list_usb_performax_devices())

stage = Arcus.GenericPerformaxStage() # this alternated with the other 2`

What I see as return is: `[(0, 'jsa00', 'Performax USB', '\\\\?\\usb#vid_1589&pid_a101#jsa00#{3c5e1462-5695-4e18-876b-f3f3d08aaf18}', '1589', 'a101')]`

and then:
`Traceback (most recent call last):
  File "c:\simplifiedPath\DMXcontrolTest.py", line 15, in <module>
    stage = Arcus.Performax4EXStage()
  File "C:\simplifiedPath\Python39\site-packages\pylablib\devices\Arcus\performax.py", line 119, in __init__
    super().__init__(idx=idx)
  File "C:\simplifiedPath\Python39\site-packages\pylablib\devices\Arcus\performax.py", line 50, in __init__
    self.open()
  File "C:\simplifiedPath\Python39\site-packages\pylablib\devices\Arcus\performax.py", line 68, in open
    raise ArcusError("can't connect to the stage with index {}".format(self.idx))
pylablib.devices.Arcus.ArcusPerformaxDriver_lib.ArcusError: can't connect to the stage with index 0`

I may be using it wrong, but I would welcome to learn how to do it right please!

AlexShkarin, 2021-11-08:
Hello!

It looks like the stage is used by some other software. It can be, for example, the manufacturer control software, another Python interpreter, or even the same Python console if you tried to open the stage before and it failed. Therefore, I would suggest closing all Python interpreters and start a fresh one to make sure that the device is not being held up anywhere else.
If that does not help, can you confirm that you can control the stage with its native software, if you have any available?

I have also briefly looked at the DMX-J-SA-17 documentation, and it seems like its communication protocol is different enough from PMX-2EX/4EX that it would not work right away. Hence, I would suggest using only ``GenericPerformaxStage`` for now, since the other classes (e.g., ``Performax2EXStage``) will certainly raise an error. To make sure that the connection is established and working, you can call ``stage.query("PX")`` and see if it returns the current position.

Once this connection problem is resolved, we can look into adjusting the current PMX code to work with DMX-J-SA stages.

otamichalekSTFC, 2021-11-09:
Hi,

brilliant, it works now! Thank you and sorry for bothering you with such a simple issue! 

Now I can easily use the motor with .query and DMX-J-SA ASCII commands, which is perfect for my task.

Thank you!

AlexShkarin, 2021-11-09:
Great! If you are comfortable with the ASCII command interface, then this ``GenericPerformaxStage`` class is indeed all you should need.

At his point I might want to add this device to one of the next releases of the library. When I do, I would be grateful if you could check it with your motor and confirm that everything works (writing classes just based on the documentation does not always work, so it's better to check it on the real hardware whenever possible).

otamichalekSTFC, 2021-11-10:
Yes, thank you!

I would be happy to, let me know when you have it ready!

AlexShkarin, 2021-12-12:
Hi!

Just to follow up on that earlier request: the control for the motor is now added to the library (starting from v1.3.0) as ``PerformaxDMXJSAStage``. I would appreciate if at some point you could quickly check if it works.

Sincerely,
Alexey.

otamichalekSTFC, 2021-12-13:
Hi,

I tested a few basic commands on Arcus DMX-J-SA-17 from [here](https://pylablib.readthedocs.io/en/latest/devices/stages_basics.html) and they seem to work ok.

Namely:
`move_by, jog, set_axis_speed, stop, close, wait_for_home, home, wait_move, is_moving, get_position, set_position_reference, move_to`

I think we can close this issue for now unless I find any problems in the future when using it.

Thank you very much!