AlexShkarin/pyLabLib #4: Funciton 'GetPreAmpGain' raised error 20066(DRV_P1INVALID)

whooie, 2021-11-08:
I'm trying to set up control of an Andor iXon Ultra 888 using this library and can't connect due to a `DRV_P1INVALID` error raised in the execution of this line:

https://github.com/AlexShkarin/pyLabLib/blob/f935595f20057330a58934afecc4a9d467bd2969/pylablib/devices/Andor/atmcd32d_lib.py#L660

which was called involved in the execution of

https://github.com/AlexShkarin/pyLabLib/blob/f935595f20057330a58934afecc4a9d467bd2969/pylablib/devices/Andor/AndorSDK2.py#L174

following the original connection attempt,

```Python
from pylablib.devices import Andor
cam = Andor.AndorSDK2Camera()
```

Seems like it shouldn't matter, but just to be thorough this is using a `pylablib-lightweight[devio,gui-pyqt5]` installation on Windows 10.

AlexShkarin, 2021-11-09:
Could you check your Andor SDK2 version?

Accroding to its changelog, version 2.94.30001.0 fixed bug stating ``Fix for SDK defaulting to using an unavailable PAG for iXon 888``. That could be the reason you're experiencing this issue.

If updating SDK2 version does not help, you can try moving ``preamp_gain=self.GetPreAmpGain(pa)`` (line 660 in ``atmcd32d_lib.py``) inside the ``try`` block immediately following it, i.e., move it one line down and add an extra indent. This should get rid of this particular error, although there might be similar error further down the line (e.g., when trying to get or set the preamp gain or the amplifier mode). If you do try this, could you also report the result of executing ``cam.get_all_amp_modes()``?

whooie, 2021-11-09:
I'm using SDK version 2.104.33000.0.

I tried making that change in `atmcd32d_lib.py` and got `DRV_NOT_SUPPORTED` from `SetNumberPrescans`. Here's the traceback:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Covey Lab\pyLabLib\pylablib\devices\Andor\AndorSDK2.py", line 112, in __init__
    self.open()
  File "C:\Users\Covey Lab\pyLabLib\pylablib\devices\Andor\AndorSDK2.py", line 239, in open
    self._setup_default_settings()
  File "C:\Users\Covey Lab\pyLabLib\pylablib\devices\Andor\AndorSDK2.py", line 182, in _setup_default_settings
    self.setup_kinetic_mode(1)
  File "C:\Users\Covey Lab\pyLabLib\pylablib\devices\Andor\AndorSDK2.py", line 69, in wrapped
    res=func(self,*args,**kwargs)
  File "C:\Users\Covey Lab\pyLabLib\pylablib\devices\Andor\AndorSDK2.py", line 704, in setup_kinetic_mode
    lib.SetNumberPrescans(num_prescan)
  File "<string>", line 1, in <lambda>
  File "C:\Users\Covey Lab\pyLabLib\pylablib\core\utils\ctypes_wrap.py", line 278, in wrapped_func
    retval=func(*call_args)
  File "C:\Users\Covey Lab\pyLabLib\pylablib\devices\Andor\atmcd32d_lib.py", line 41, in errchecker
    raise AndorSDK2LibError(func.__name__,result)
pylablib.devices.Andor.atmcd32d_lib.AndorSDK2LibError: function 'SetNumberPrescans' raised error 20991(DRV_NOT_SUPPORTED)
```

Also, here is the output from `get_all_amp_modes` (I printed them from the `AndorSDK2Lib` call in `atmcd32d_lib.py` since the camera didn't get fully set up):
```
[TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Electron Multiplying', hsspeed=0, hsspeed_MHz=30.0, preamp=0, preamp_gain=1.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Electron Multiplying', hsspeed=0, hsspeed_MHz=30.0, preamp=1, preamp_gain=2.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Electron Multiplying', hsspeed=1, hsspeed_MHz=20.0, preamp=0, preamp_gain=1.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Electron Multiplying', hsspeed=1, hsspeed_MHz=20.0, preamp=1, preamp_gain=2.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Electron Multiplying', hsspeed=2, hsspeed_MHz=10.0, preamp=0, preamp_gain=1.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Electron Multiplying', hsspeed=2, hsspeed_MHz=10.0, preamp=1, preamp_gain=2.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='ElectronMultiplying', hsspeed=3, hsspeed_MHz=1.0, preamp=0, preamp_gain=1.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Electron Multiplying', hsspeed=3, hsspeed_MHz=1.0, preamp=1, preamp_gain=2.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=1, oamp_kind='Conventional', hsspeed=0, hsspeed_MHz=1.0, preamp=0, preamp_gain=1.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=1, oamp_kind='Conventional', hsspeed=0, hsspeed_MHz=1.0, preamp=1, preamp_gain=2.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=1, oamp_kind='Conventional', hsspeed=1, hsspeed_MHz=0.10000000149011612, preamp=0, preamp_gain=1.0), TAmpModeFull(channel=0, channel_bitdepth=16, oamp=1, oamp_kind='Conventional', hsspeed=1, hsspeed_MHz=0.10000000149011612, preamp=1, preamp_gain=2.0)]
```

AlexShkarin, 2021-11-09:
OK, it looks like there are still some capabilities which are not always supported, but which the class does not check. Could you replace the line 704 in AndorSDK2.py (it's the ``lib.SetNumberPrescans(num_prescan)`` line which raises the error) with
```python
        if self._check_option("set",AC_SETFUNC.AC_SETFUNCTION_PRESCANS):
            lib.SetNumberPrescans(num_prescan)
        else:
            num_prescan=0
```
I apologize for such awkward step-by-step debugging, but the documentation is not always 100% clear on which errors can arise in which situation, and which functions may or may not be available.

whooie, 2021-11-09:
That seems to work -- at least no exceptions are thrown now when I connect to the camera. I'm still in the process of setting everything up so I don't have any tests I can run to make sure it fully works, but I'll let you know if I encounter anything else. Thanks!