AlexShkarin/pyLabLib #5: KDC 101 Error Backend exception: D2XXException(3) ('Device Not Opened')

Habubay, 2021-12-03:
Hello, 

Firstly, thanks a lot for the great work done on this. Stumbled across it while trying to automate certain thorlabs stages and I believe it will be very useful. 

I am running into issues with connecting the stage. I am able to detect the controller (KDC101 - Brushed Motor Controller) that is connected. 

[image]

However, when I connect to the stage, I get the following error 

[image]

[image]

Please any insights into why I am getting this error will be appreciated.

AlexShkarin, 2021-12-03:
Hello!

This error should not normally show up, and I have not personally seen it before. It seems like it might be realted to problems with FTDI drivers, especially on Linux or MacOS. For example, such error is discussed [here](https://stackoverflow.com/questions/28461433/unable-to-open-a-connection-to-an-ftdi-device) or [here](https://github.com/thomaschaaf/node-ftdi/issues/4). Some other reports [here](https://www.edaboard.com/threads/problem-with-d2xx-as-it-cant-open-jtag.127837/) or [here](https://forums.raspberrypi.com/viewtopic.php?t=159181) also suggest that it might be related to permissions (i.e., you need to run the code as root). Unfortunately, I could not find any more information so far.

In case you're using Windows, I have one more suggestion: can you install [Thorlabs APT software](https://www.thorlabs.de/software_pages/ViewSoftwarePage.cfm?Code=Motion_Control&viewtab=1) and confirm that you can connect to the stage from there?

Sincerely,

Alexey.

Habubay, 2021-12-04:
Hi Alex, 

Thanks I was able to rectify it. After connecting to the stage once and not closing that connection, re attempting to connect leads to that error. I now have stage.close() in all my code blocks and that seems to solve the problem. 

Thanks a lot for the suggestions and helping to look into this :).

closed 2021-12-05 by Habubay