AlexShkarin/pyLabLib #7: Function serial_readline() fails with error about type mismatch

voslik, 2021-12-17:
When I call the function serial_readline() to read bytes off of my camera, I get an error about mismatched types. As best as I can tell, it seems to me that the wrapper function is checking whether pv is of type c_char_p (it is not, it is c_char_Array) and then tries to convert it to a c_char_p, which fails.

**My code:**

from pylablib.devices import IMAQ

IMAQ.list_cameras()
cam1 = IMAQ.IMAQCamera('img0.iid')
cam1.setup_serial_params(write_term='\r', datatype='str')

msg_transmitted = '*IDN?'
error_message = cam1.serial_write(msg_transmitted,timeout=1.0, term=None)

msg_received = cam1.serial_readline(timeout=1.0, datatype=None, maxn=1024)

**Error output:**

Traceback (most recent call last):

  File "C:\Users\admin\Desktop\my_folder\test.py", line 28, in <module>
    msg_received = cam1.serial_readline(timeout=1.0, datatype=None, maxn=1024)

  File "C:\Users\admin\anaconda3\lib\site-packages\pylablib\devices\IMAQ\IMAQ.py", line 397, in serial_readline
    msg,_=lib.imgSessionSerialRead(self.sid,maxn,int(timeout*1000))

  File "<string>", line 1, in <lambda>

  File "C:\Users\admin\anaconda3\lib\site-packages\pylablib\core\utils\ctypes_wrap.py", line 271, in wrapped_func
    func_args=self._prepare_arguments(argnames,prep_argtypes,kwargs,argprep)

  File "C:\Users\admin\anaconda3\lib\site-packages\pylablib\core\utils\ctypes_wrap.py", line 175, in _prepare_arguments
    cv=pv if isinstance(pv,t) else t(pv)

TypeError: bytes or integer address expected instead of c_char_Array_1025 instance

AlexShkarin, 2021-12-17:
You are right, there appears to be a type mismatch. You can fix it by replacing the two following lines inside `site-packages/pylablib/devices/IMAQ/niimaq_lib.py`:

- Line `98` from
```python
argprep={"buffer":lambda bufSize:ctypes.create_string_buffer(bufSize+1)}, byref=["bufSize"])
```
to
```python
argprep={"buffer":lambda bufSize:ctypes_wrap.strprep(bufSize+1)()}, byref=["bufSize"])
```
- Line `101` from
```python
argprep={"buffer":lambda bufferSize:ctypes.create_string_buffer(bufferSize+1)}, byref=["bufferSize"])
```
to
```python
argprep={"buffer":lambda bufferSize:ctypes_wrap.strprep(bufferSize+1)()}, byref=["bufferSize"])
```
What it does is replacing ``ctypes.create_string_buffer(l)`` with ``ctypes_wrap.strprep(l)()``, which does the same buffer allocation, but casts it to ``c_char_p`` afterwards.
Could you check if this solves the issue? If it does, I will implement it in the next release.

Sincerely,

Alexey.

voslik, 2021-12-20:
Yes, that worked. Thank you.

closed 2022-02-11 by AlexShkarin