AlexShkarin/pyLabLib #8: ANDOR iVac connection problem

sarahrahsl, 2022-01-06:
Hello! 

Thank you for creating this package! I started using the pyLabLib to control my ANDOR iVac 316 detector few days ago, and had successfully acquired signals with the acquisition settings I input. However, after a while on the same day, the spectrometer returned an array of zeros when I run the exact same code. Therefore, I switch back to using the ANDOR Solis software to check if the spectrometer is responding or not. When I was using the software by ANDOR, the connection between the software and the device was fine, the cooling function worked, except it kept returning an array of zeros no matter what settings I made. I also tried to call the device through LABVIEW (which had worked previously), but it also had the same problem despite the connection was fine. This problem persists ever since I run the following few codes: 

    from pylablib.devices import Andor
    import pylablib as pll
    pll.par["devices/dlls/andor_sdk2"] = "D:/Andor SDK"
    Andor.get_cameras_number_SDK2()
    cam = Andor.AndorSDK2.AndorSDK2Camera()
    cam.set_read_mode("fvb")
    cam.set_exposure(0.1) # set exposure to 100ms
    cam.start_acquisition()
    frame = cam.snap()
    cam.stop_acquisition()
    cam.close()

Restarting the computer or reinstalling the SDK (version 2.103.30022.0) doesn't work. At this point I am wondering if it is because I used the function incorrectly, causing the hardware to malfunction or not responding anymore. But since it returns a array of zeros, I don't think the sensors inside the spectrometer are damaged, it is more like the shutter is closed and no light is detected. Have you encountered any situation like this? Would be grateful if you could look into this issue for me, thanks!

AlexShkarin, 2022-01-06:
Hello!

Unfortunately, I have not encountered such problems before. I do not have a lot of experience with spectrometer cameras: I briefly used the library code on a Newton camera, but that's about it. However, I have been using it with Andor iXon for several years, and I have not seen such a sudden loss of signal before. Was there anything specific that prompted it? Have you observed any strange behavior immediately prior to this?

I can suggest a couple of things to check, although I can't guarantee that any of them work. To avoid possible complications from pylablib code, I would do all of these tests in Andor Solis using your standard settings.

- In general, I'm surprised that you get exactly zeros for all pixels. In my experience, this does not normally happen when the chip is dark; you should still see some dark noise. Hence, I doubt that it's something as simple as shutter settings.
- That being said, you might want to make sure that the cooling is working, that the sensor temperature reaches the setpoint, and that the setpoint is the same as it used to be. Several times I've noticed that when the chip is warm (usually, the first couple of minutes after the cooling is on), the sensitivity drops and the frame becomes completely zero. In addition, you can still check the shutter settings and ensure that it's not always closed.
- Sometimes one might get zeros when the camera is extremely saturated, so I'd recommend checking that there's no strong light landing on the chip. Furthermore, you can try switching into the regular image mode instead of the full vertical binning (FVB) to check if the whole image is still all zeros.
- Related to that, you can try different shift speeds and preamp gain modes while in the full image mode, and see if it makes any difference.
- In rare cases there are some weird driver or software interactions which are not easily solved by reinstalling the software. Hence, it might help to check the camera with an entirely different PC. You should also power-cycle the camera (perhaps even several times) to make sure that its controller is not just stuck in a weird state.

sarahrahsl, 2022-01-07:
Thank you for the suggestions. I have tried switching to a completely different computer and the same thing happened, so it shouldn't be the driver's problem. I checked the shutter, and tried switching different parameters for acquisition setting as well, but it didn't help. I think a more possible reason is the second point you mentioned about the cooling, since I did use the library to call the cooler again and again, setting different temperatures to try out the code... Although after that the temperature indicator still showed the temperature has reached the setpoint when I tried it on Solis software, I am not sure if it has actually responded to the temperature setting, because I am not able to check the interior of the detector, and there aren't any noise or heat from the detector when it is actually working. So maybe I did some improper operations to the detector when I was trying out the code and that has caused to some hardware problems. I am double checking with the manufacturer as well to solve this problem and I will keep you posted. Thank you for helping me even this might not be a problem of the code!

AlexShkarin, 2022-02-11:
Hello!

I was wondering if you have heard anything from Andor. Were you able to figure out the issue?

sarahrahsl, 2022-02-11:
Hello, thanks for following up!

I contacted the technician, ran through some tests and they suggested sending it back for repair. So the ANDOR camera was sent back to the factory, still haven't heard back from them. Will let you know what the problem was when I receive further update from them!

(Also I recently I used python to call labview to control another spectrometer from Thorlabs, which I encountered the same situation, but this time it could easily solved by unplugging and replugging the usb connection. Don't know if the two issues are related to the same bug. Anyways, I will wait for their reply)

AlexShkarin, 2022-02-11:
Thank you for the update! I hope the issue will be resolved soon!

sarahrahsl, 2022-05-12:
Hi again! I have finally received the diagnostic report for the spectrometer after such a long delay. The report stated that the electronic component failure has made the spectrometer not functioning, and they fit a new PCB for the spectrometer. Beside that they didn't really say much about how to prevent these kind of situations and what has caused it.. anyway, I will be more careful when I use the library to control the spectrometer. Thanks!

AlexShkarin, 2022-05-20:
Thank you for letting me know of the results! I'm glad it's working again. Hopefully, the issue was not caused by the software and won't repeat in the future.

closed 2022-05-20 by AlexShkarin