AlexShkarin/pyLabLib #10: Thorlabs Kinesis enable_channel() Function

Em607867, 2022-01-21:
Alex, 

Just wanted to bring something I noticed to your attention. 

The _enable_channel()_ function which enables or disables a given channel returns an error message which I'm guessing you're aware of based on the comment '# seems not to work for most devices when tested' below; 

[image]

However, this actually works as intended on the K-cube (KBD 101) controller (i.e it actually disables or enables the channel) even though this error message pops up. Wonder if there is a feedback on the backend (recv_comm to check whether channel is enabled or not) that is causing this error. 

On my end this isn't a problem as I can ignore the error message but thought it's something worth sharing. 

Thanks again for your great work. 

Best Regards

AlexShkarin, 2022-01-24:
Dear Manny,

Thanks for pointing it out! It looks like I used ``self.query`` instead of ``self.send_comm`` inside the ``_enable_channel`` method, so the code expects a reply which never comes. It will be fixed in the new release.

There is, however, another error which I have seen occasionally (e.g., with a TDC101 device), where ``_is_channel_enabled`` would always return ``False``, although ``_enable_channel`` still functions. Could you please check if it works with your controller?

Sincerely,

Alexey.

Em607867, 2022-01-25:
ap Hi Alexey, 

I have also previously noticed the issue you described with _enable_channel function. I get an error message below even though the channel is enabled: 

[image]