AlexShkarin/pyLabLib #18: Thorlabs UC480 Camera snap/grab image sizes

dhblight, 2022-06-07:
Hi, I'm using a throlabs UC480 Camera but am struggling to get the right sized image from the grab/snap functions. The camera is 1024x1280 pixels but the data is returned as 1280x3. Using the _read_buffer() function appears to return the full data:

[image]

[image]

[image]

Is there any way to get the 1024x1280 image from grab/snap or do you know the file those functions are taken from as i can't seem to find it?

AlexShkarin, 2022-06-07:
Hi!

It looks like there is an interaction between the color channels and the chunk readout code used in some cameras for speedup. As far as I can see, there are two solutions for you right now:

1. If you are fine with working in a monochrome mode, you can use ``set_color_mode`` method to set the readout to monochrome, e.g., ``cam.set_color_mode("mono8")``. If that mode does not work, you can check all available color modes using ``cam.get_all_color_modes()`` and see if any of them starts with ``mono``.
2. I implemented a fix, but it might be several weeks until I release the next pylablib version to make it available on PyPi. If you want, you can look at the commit 40785a66898812eb23d1e75e752ce68d48432ffc and manually replace the two modified files in your Python ``Lib\site-packages\pylablib`` folder. More specifically [camera.py](https://raw.githubusercontent.com/AlexShkarin/pyLabLib/40785a66898812eb23d1e75e752ce68d48432ffc/pylablib/devices/interface/camera.py) goes into  ``Lib\site-packages\pylablib\devices\interface\camera.py`` and [uc480.py](https://raw.githubusercontent.com/AlexShkarin/pyLabLib/40785a66898812eb23d1e75e752ce68d48432ffc/pylablib/devices/uc480/uc480.py) replaces  ``Lib\site-packages\pylablib\devices\uc480\uc480.py``

Let me know if any of this works.

dhblight, 2022-06-07:
Hi Alex,

Thank you so much! I manually replaced the 2 files and snap and grab are both now returning the whole image.

[image]

closed 2022-06-07 by dhblight