AlexShkarin/pyLabLib #20: plotting of data

tvandersar, 2022-06-17:
Hi Alex, 

first of all, very impressive work on pylablib! I just discovered it and the communication with instruments is very easy to use. I am currently using matlab in my lab but am running into slow GUIs and other difficulties and would like to switch to a python-based measurement framework. I had a look at qcodes too, but it seems the driver database of pylablib is really great. I got the NiDAQ card running and Andor camera running in no time.

I was hoping you could tell me if you have included functionality for live plotting of data. For instance, I would like to see the counts gathered by a NiDAQ card plotted in a live graph inside a gui that can also be used to set measurement parameters. Do you have any pointers on how to start this?

AlexShkarin, 2022-06-20:
Hi! I'm glad you find the library useful!

I have some basic plotting functionality, but it is not very well documented, so I would not recommend using it at the moment. In any case, it is simply a wrapper around [pyqtgraph](https://www.pyqtgraph.org/), which is a go-to library for real-time plotting. 

The way I implement GUI myself is by using PyQt/PySide for a general layout and pyqtgraph for plotting. The whole topy is quite large, so I can not describe anything in details here. I'd recommend to start with [ZetCode](https://zetcode.com/gui/pyqt5/) tutorial for the general GUI application introduction, maybe followed by some [generic tutorials](https://wiki.python.org/moin/PyQt/Tutorials). One suggestion I would make is to use Qt5 or Qt6 instead of Qt4 (too old) unless you have some specific reasons to do otherwise.

tvandersar, 2022-06-21:
Thank you for your quick answer! I have been reading your code this weekend and understood better how you created the camera-control gui. In particular the nice programmatic ways of making various gui controls. 

I was also looking into qCodes. Its level of abstraction might be useful (with stations, experiments, datasets) but its focus on electronic transport measurements is different from our optics lab and is visible in the driver set. Not sure what package to go to from here. I would love to hear your thoughts on it but also understand if you are busy.

I saw you were considering co-development of code in a question raised by another user. Our measurements on NV centers in diamond are related to the single-molecule work, so it might be a good fit. In any case I plan to keep trying out pyliblab!