AlexShkarin/pyLabLib #23: Andor IDus does not connect

1234bnm, 2022-11-02:
Hey, 

I am currently trying to set up a Spectrometer. 
for that I am using an older Andor Idus 490. 

If I use the function Andor.get_cameras_number_SDK2() the feedback is 1 so I guess the implimentation of the dll worked 

But if I try to connect the cam via cam = Andor.AndorSDK2Camera() 

I get the following Error: 

AndorSDK2LibError: function 'GetFastestRecommendedVSSpeed' raised error 20991(DRV_NOT_SUPPORTED) 

is there a way I can solve this error? 

best regards

AlexShkarin, 2022-11-02:
Hello!

It looks like some of the standard functions are not supported by the camera, but the code does not account for it and raises an exception. Could you try and replace the file ``Lib\site-packages\pylablib\devices\Andor\AndorSDK2.py`` within your Python installation with the [modified one](https://raw.githubusercontent.com/AlexShkarin/pyLabLib/2b6461228e7238ad96652957d6c0a13cbe3d0f69/pylablib/devices/Andor/AndorSDK2.py) and let me know if it fixes the issue?

1234bnm, 2022-11-09:
Hello, 

that worked like a charm; thank you verry much. 

I am ontro the next error that has to do with the firmware I am gonna try to find a way to update it and let you know if I am getting any pictures. 

best regards

1234bnm, 2022-11-23:
Hello Alex, 

I got the firmware updated (at least I thinks so) and now I am having a different Problem. 

right now I am getting the Error : 

AndorSDK2LibError: function 'SetMultiTrack' raised error 20068(DRV_P3INVALID)

when I want to initialize the Camera; do you have any idea how I can fix that. 

Thanks fot the help!

AlexShkarin, 2022-11-23:
Looks like a smilar problem: the camera does not support the "default" multi-track options set up during the camera connection. I would suggest opening the same ``AndorSDK2.py`` file (located at ``Lib\site-packages\pylablib\devices\Andor\AndorSDK2.py``) and changing line 870 from
```python
def setup_multi_track_mode(self, number=1, height=1, offset=1):
```
to 
```python
def setup_multi_track_mode(self, number=1, height=1, offset=0):
```
i.e., change the default offset value. In addition, if that works, could you also let me know what sensor size you get from the camera, i.e., what's the result of calling ``cam.get_detector_size()``?

1234bnm, 2022-11-30:
Hello Alex, 

It does connect now and if I run cam.get_detector_size() the output is (512,1) which is what I was expecting. 

Thank you verry much.

Moronta, 2022-12-21:
Hello,

I  also have iDus camera (DU401A-BV) but in my case the initialization has troubles with the shutter.. I do not have an external shutter connected. Did this happen to you @1234bnm ?

You will find the output errors below,

Thank you very much,

Pedro.

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py", line 113, in __init__
    self.open()
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py", line 242, in open   
    self._setup_default_settings()
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py", line 178, in _setup_default_settings
    #            self.setup_shutter("closed")
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py", line 70, in wrapped 
    res=func(self,*args,**kwargs)
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\core\devio\interface.py", line 666, in wrapped   
    res=func(**all_args)
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py", line 543, in setup_shutter
    #        min_open_time,min_close_time=self.get_min_shutter_times()
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py", line 70, in wrapped 
    res=func(self,*args,**kwargs)
  File "C:\Users\LAB_045_1\Documents\PedroM\dev\SLM\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py", line 531, in get_min_shutter_times
    #        return lib.GetShutterMinTimes()
TypeError: 'NoneType' object is not callable

1234bnm, 2022-12-21:
Hello Pedro, 

this error did not happen, while I was working with the camera.

AlexShkarin, 2022-12-21:
Dear Pedro,

This error most likely means that the function ``GetShutterMinTimes`` is not defined in the Andor DLL. According to the documentation, it was introduced in version 2.93.300000.0, so most likely you just have an earlier version. To check it, you can run Andor Solis and go to ``Help -> About...`` in the top menu. It will be there under SDK Version 2 (e.g., 2.92.30008.0).

To fix the problem, you can replace line 531, which currently reads
```python
return lib.GetShutterMinTimes()
```
with
```python
return lib.GetShutterMinTimes() if lib.GetShutterMinTimes is not None else (0,0)
```
Let me know if this solve the problem, and if there are any other issues.

Sincerely,

Alexey.