AlexShkarin/pyLabLib #27: Kinesis Piezo Motor Channel 3 not Working

LJJS, 2022-12-12:
Hi,
The channel 3 with a KIM101 Piezo Controller is not working properly as it switched automaticlly to channel 4.

In the following example the channel 3 is enabled correctly at the KIM101. However, the channel is not moving and the "move_by" function seems to have no effect.

```
from pylablib.devices import Thorlabs
stage = Thorlabs.KinesisPiezoMotor("XXXXXXXX")

stage.enable_channels((3,))
print(stage.get_enabled_channels())
stage.move_by(1000, auto_enable=False, channel=3)

stage.close()
```

When "auto_enable" is `True`, the channel 3 will be enable in the line above for a short moment (also at the device), but the "move_by" function switches to channel 4 and channel 4 is moving instead.

```
from pylablib.devices import Thorlabs
stage = Thorlabs.KinesisPiezoMotor("XXXXXXXX")

stage.enable_channels((3,))
print(stage.get_enabled_channels())
stage.move_by(1000, channel=3)

stage.close()
```

I had a quick look at the implementation, however I think the problem is in some Hex commands sent to the KIM101 but I am not sure.

All the other channels work properly, just channel 3 switched automaticlly to channel 4 every time.

AlexShkarin, 2022-12-12:
Hello!

Thanks for pointing it out! It looks like there's a bug specifically in the ``move_by`` implementation, and ``move_to`` should work fine. So one option can be using ``stage.move_to(stage.get_position(channel=3)+distance, channel=3)`` instead of ``stage.move_by(distance, channel=3)`` (which is how it's implemented in the code anyway).

To fix the bug, you can remove lines 859 (``@interface.use_parameters(channel="channel_id")``) and 861 (``self._pzmot_autoenable(channel,auto_enable)``) from ``kinesis.py`` After that it should work fine.

Let me know if it works.

Sincerely,

Alexey.

LJJS, 2022-12-13:
Hi,
Thanks for the quick response!
I tried your solutions:

1) The `move_to` workaround works perfectly!

2) In the `kinesis.py` file I deleted the lines you mentioned (but with other line numbers 865 and 868) for the `_pzmot_move_by` function and it also worked just fine!