AlexShkarin/pyLabLib #28: KIM101

drbel8, 2022-12-21:
Hi, 
from a Linux environment I'm trying to connect to a KIM101 but 
In [1]: from pylablib.devices import Thorlabs
In [2]: Thorlabs.list_kinesis_devices()

results in
Segmentation fault (core dumped)

while 
In [1]: from pylablib.devices import Thorlabs
In [2]: x = Thorlabs.KinesisPiezoMotor("XXXXXXXX")
in a LibFtdiException

I've seen from another bug (#27 ) another user is able to use a KIM101, but I didn't want to "pollute" the discussion.

Any suggestion on how to move forward?

AlexShkarin, 2022-12-21:
Hi!

Unfortunately, I have limited exprience with device communications in Linux, so I can't tell what exactly the issue is.

First thing would be to figure out if the device is recognized by the OS at all, and how it is addressed. As far as I undestand, Thorlabs APT devices (such as KIM101, KDC101, or FW102) might show up directly as a serial port, e.g., ``dev/ttyUSB0``. Could you check that? For example, if there's an analog of Windows device manager, you can unplug the device and plug it back in, and check what changes.

AlexShkarin, 2022-12-22:
Ok, in this case could you try giving this address to the device class instead of its ID? That is, execute ``x = Thorlabs.KinesisPiezoMotor("/dev/ttyUSB0")`` In principle, the code should work with the standard serial interface library, though I haven't tested it in this more.

drbel8, 2022-12-22:
Thanks for your help, I copy and paste the Exception (it's identical to the one I get with the serial number)

In [1]: from pylablib.devices import Thorlabs
In [2]: x = Thorlabs.KinesisPiezoMotor("/dev/ttyUSB0")
---------------------------------------------------------------------------
LibFtdiException                          Traceback (most recent call last)
~/.local/lib/python3.10/site-packages/pylablib/core/devio/comm_backend.py in __init__(self, conn, timeout, term_write, term_read, open_retry_times, datatype, reraise_error)
    805             try:
--> 806                 self.instr=self._open_instr(port,conn_dict)
    807                 self.opened=True

~/.local/lib/python3.10/site-packages/pylablib/core/devio/comm_backend.py in _open_instr(self, port, params)
    817             if "serial_number" in sig.arg_names: # pyft232 v0.11 signature change
--> 818                 return ft232.Ft232(serial_number=port,**params)
    819             else:

~/.local/lib/python3.10/site-packages/ft232/libftdi.py in __init__(self, port, serial_number, description, baudrate, bytesize, parity, stopbits, timeout, xonxoff, rtscts, writeTimeout)
    163                                           PRODUCT, None, serial)
--> 164             if ret != 0: raise LibFtdiException(self._context)
    165         elif description:

<class 'str'>: (<class 'TypeError'>, TypeError('__str__ returned non-string (type bytes)'))

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
<ipython-input-2-a3ec515f9cd1> in <module>
----> 1 x = Thorlabs.KinesisPiezoMotor("/dev/ttyUSB0")

~/.local/lib/python3.10/site-packages/pylablib/devices/Thorlabs/kinesis.py in __init__(self, conn, default_channel)
   1215     """
   1216     def __init__(self, conn, default_channel=1):
-> 1217         super().__init__(conn,default_channel=default_channel)
   1218         self.add_background_comm(0x08D6) # move completed
   1219         self._add_status_variable("enabled_channels",self.get_enabled_channels,self.enable_channels)

~/.local/lib/python3.10/site-packages/pylablib/devices/Thorlabs/kinesis.py in __init__(self, conn, timeout, default_channel, is_rack_system)
    250     """
    251     def __init__(self, conn, timeout=3., default_channel=1, is_rack_system=False):
--> 252         super().__init__(conn,timeout=timeout,is_rack_system=is_rack_system,default_axis=default_channel)
    253         self._remove_device_variable("axes")
    254         self._add_info_variable("channel",self.get_all_channels)

~/.local/lib/python3.10/site-packages/pylablib/devices/interface/stage.py in __init__(self, default_axis, *args, **kwargs)
     32     _axis_value_case=None
     33     def __init__(self, *args, default_axis="all", **kwargs):
---> 34         super().__init__(*args,**kwargs)
     35         self._original_axis_parameter=None
     36         self._default_axis=default_axis

~/.local/lib/python3.10/site-packages/pylablib/devices/Thorlabs/kinesis.py in __init__(self, conn, timeout, is_rack_system)
     42     def __init__(self, conn, timeout=3., is_rack_system=False):
     43         defaults={"serial":{"baudrate":115200,"rtscts":True}, "ft232":{"baudrate":115200,"rtscts":True}}
---> 44         instr=comm_backend.new_backend(conn,backend=("auto","ft232"),term_write=b"",term_read=b"",timeout=timeout,
     45             defaults=defaults,reraise_error=ThorlabsBackendError)
     46         instr.setup_cooldown(write=0.003)

~/.local/lib/python3.10/site-packages/pylablib/core/devio/comm_backend.py in new_backend(conn, backend, defaults, **kwargs)
   1567     if defaults is not None and backend_name is not None and backend_name in defaults:
   1568         conn=backend.combine_conn(conn,defaults[backend_name])
-> 1569     return backend(conn,**kwargs)
   1570 def backend_error(backend, conn=None):
   1571     """

~/.local/lib/python3.10/site-packages/pylablib/core/devio/comm_backend.py in __init__(self, conn, timeout, term_write, term_read, open_retry_times, datatype, reraise_error)
    811                 self._conn_params=(port,conn_dict,timeout)
    812             except self.BackendError as e:
--> 813                 raise self.Error(e) from e
    814 
    815         def _open_instr(self, port, params):

~/.local/lib/python3.10/site-packages/pylablib/core/devio/comm_backend.py in __init__(self, exc)
     21     """Generic exception relaying a backend error"""
     22     def __init__(self, exc):
---> 23         msg="backend exception: {} ('{}')".format(repr(exc),str(exc))
     24         super().__init__(msg)
     25         self.backend_exc=exc

TypeError: __str__ returned non-string (type bytes)

AlexShkarin, 2022-12-22:
Ah, looks like there's an error in a regex, so it doesn't recognize that it needs to use serial backend. Could you try ``x = Thorlabs.KinesisPiezoMotor(("serial","/dev/ttyUSB0"))`` instead? This specifies the backend explicitly.

drbel8, 2022-12-22:
we are getting closer,

ThorlabsBackendError: backend exception: 'read returned less than expected: 0 instead of 6' ('read returned less than expected: 0 instead of 6')

AlexShkarin, 2022-12-22:
Ok, looks like there's another minor bug regarding default backend settings. Could you now try
```python
conn = {"port":"/dev/ttyUSB0","baudrate":115200,"rtscts":True}
x = Thorlabs.KinesisPiezoMotor(("serial",conn))
```

drbel8, 2022-12-22:
success! :)
thank you very much for your help and your time

AlexShkarin, 2022-12-22:
Great! Thanks for testing the code for me!

In the next release I'll update the documentation and fix these two bugs, so that the simple ``Thorlabs.KinesisPiezoMotor("/dev/ttyUSB0")`` will also be able to work.

drbel8, 2022-12-22:
thanks again!

closed 2022-12-22 by drbel8