AlexShkarin/pyLabLib #29: Thorlabs Kinesis - MPCX20 Issues

zac-castillo, 2023-01-23:
Hello,

I'm attempting to use the kinesis package to control a Thorlabs MPC220 controller. I've been able to connect and perform basic movements for a single channel using the `move_to` method, but many (useful) commands return errors. 

For example, the methods `home`, `is_moving`, `wait_move`, and `wait_for_stop `all return the error:
`ThorlabsError: unexpected command received: expected 0x042a, got 0x0491`

The `get_status` and `get_status_n` methods return the error:
`ThorlabsError: unexpected command received: expected 0x042a, got 0x0491`

The `get_device_info()` method returns the error:
`ThorlabsError: unexpected command received: expected 0x0006, got 0x0491`

I can provide more examples of errors, but I think the common issue might be the MPC220 and MPC320 having a different message set than the class is expecting. 

Do you have any recommendations of how to accommodate these types of  motors?

Thank you,
-Zac

AlexShkarin, 2023-01-23:
Dear Zac,

I haven't worked with these devices, so I can only speculate. It is possible that this kind of controllers occasionally sends an update message ``0x0491``, which comes instead of the expect reply and throws the rest of the code off. To check that, you should call ``dev.add_background_comm(0x0491)`` right after opening the connection (i.e., creating the device object) to indicate that such messages should be ignored. It might fix the issue completely or only partially. For example ``get_device_info()`` might work, while status-related methods like ``get_status()`` or ``is_moving()`` might still raise an error. Different outcomes will indicate whether such a simple patch is sufficient, or if one would need to add a bit more code to communicate using different status commands.

Sincerely,

Alexey.

zac-castillo, 2023-01-24:
Hi Alexey,

Thank you very much for the quick response and help. 

`get_device_info()` now works and returns the device information after calling `add_background_comm(0x0491)` right after opening the connection. However, `get_status()`, `is_moving()`, and several other methods now return the error:
`ThorlabsBackendError: backend exception: 'read returned less data than expected' ('read returned less data than expected')` 

Additionally, I'm having trouble using the second axis/channel of my controller. Here are the lines I have so far 

```
# The polarization controllers are rack systems. This must be specified
stage = Thorlabs.KinesisMotor("38293204",is_rack_system=True)
# The MPC family seems to send this reply. Use this command to ignore that response.
stage.add_background_comm(0x0491) 

num_channels = stage.get_number_of_channels()
stage.set_supported_channels(channels=num_channels)
scaling = 1370/170; # There are 1370 steps over the 170 degree range
stage.move_to(50*scaling,channel=1, scale=False)
time.sleep(2.5)
stage.move_to(115*scaling,channel=2, scale=False)
```

The result is the first channel moves to 50 degrees, a pause, then the same channel moves to 115 degrees. Have I messed up the settings to call the second stage, or is this unexpected behavior?

Thank you again,
-Zac

AlexShkarin, 2023-01-24:
Dear Zac,

First, regarding status updates still not working: it seems like these particular devices (MPCx20 controllers) do not accept the command that I normally use to get status bits (``0x0429``), and need to use a different one. I've updated the code to account for that, but for now you will need to manually replace it in your Python folder. Specifically, you should replace ``Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py`` file inside your Python folder with [this one](https://raw.githubusercontent.com/AlexShkarin/pyLabLib/37e6616ace3ed721d7506dd00f4a9dbf70991cb0/pylablib/devices/Thorlabs/kinesis.py). Ideally, the code will now automatically detect based on the device model number that it needs to use the different command. If it does not happen, you can try explicitly calling ``stage._set_status_comm(0x0490)``. This code should also take care of the periodic updates issue, so you no longer need the ``stage.add_background_comm(0x0491)`` line. In fact, you need to remove it, since otherwise the code will ignore the status reply, leading to a timeout.

Second, regarding the channels mixup. It seems to be related to the channel communication differences between the normal USB devices (which is the majority of Thorlabs devices) and rack systems. According to the documentation and the Kinesis simulation of MPCx20 devices, they are of the standard USB kind, and therefore you should not need to supply ``is_rack_system=True`` upon creation. Are you sure that the device you use is a rack system? What if yout try just
```python
stage = Thorlabs.KinesisMotor("38293204")
scaling = 1370/170; # There are 1370 steps over the 170 degree range
stage.move_to(50*scaling,channel=1, scale=False)
time.sleep(2.5)
stage.move_to(115*scaling,channel=2, scale=False)
```
and see what happens?

Sincerely,

Alexey.

zac-castillo, 2023-01-27:
Dear Alexey,

I've updated the kinesis file and  have removed `stage.add_background_comm(0x0491)` during stage initialization. I can now call methods such as `get_status()`, `get_scale()`, and `get_stage()`.  However, some methods still cause an error related to the background command `0x0491`. 
Specifically, I've found that `setup_velocity()`,  `get_velocity_parameters()`,  `setup_jog()`, `get_jog_parameters()` return the error: 
`ThorlabsError: unexpected command received: expected 0x0415, got 0x0491`
Additionally, `.get_position()` returns `ThorlabsError: unexpected command received: expected 0x0412, got 0x0491`

Thank you for pointing out that the MPCx20 devices are not rack systems. I'm not sure where I got that idea, but in any case, I can now set the position of the two channels with  `move_to()` followed by `wait_move()`.  I've tried using `move_by()`, but the controllers do not respond and no error is returned.

Ideally I would like to automate my experiments like the following:

```
scaling = 1370/170; # There are 1370 steps over the 170 degree range
angles = [0, 85, 170]

for angle1 in angles:
    stage.move_to(angle1*scaling,channel=1, scale=False)
    stage.wait_move()
    for angle2 in angles:
        
        stage.move_to(angle2*scaling,channel=2, scale=False)
        stage.wait_move()
        print(f"channel 1: {angle1}, channel 2: {angle2}")
        # Get voltage from a daq
```

I haven't tried implementing code to read a value from a daq, but I'm a bit wary since this test codes prints out all the combinations of angles before the controller has finished moving through them. Should 'wait_move()' halt the python code, or does it just cause the controller to wait executing the next command until motion is stopped?  If this is intended behavior, I suppose I could add my own bit of code to halt the code while the stage is moving.

Best regards,
-Zac

AlexShkarin, 2023-01-28:
Dear Zac,

I've looked more into the documentation, and it seems like these MPCX20 controllers are more different from other motor controllers (e.g., KDC101) than I thought, and most of the methods related to parameters (``get_velocity_parameters``, ``get_jog_parameters``, ``get_homing_parameters``, ``get_gen_move_parameters``, ``get_limit_switch_parameters``) do not apply. Instead, there is a separate command which sets up all of the polarization controller parameters: velocity, homing position, and jog size. It's not implemented yet, but it seems relatively straightforward, so I should be able to add it in the next file version.

However, there are still some puzzles in the controller behavior. First, according to the documentation ``get_position`` method should still work, so I'm a bit surprised that it fails in this particular way. Second, I would still expect ``wait_move`` method to work, since it only relies on ``get_status``. You are correct about the intending behavor: it should pause Python code execution until the controller stops moving. Hence, I would expected the code that you printed to work as intended, i.e., wait until the paddles' motion is done before printing the positions. There is only one correction you need to make: ``wait_move`` also takes the channel as an argument (which is 1 by default), so you should call ``stage.wait_move(channel=2)`` within the inner loop. You can try to see if it fixes the issue.

To figure out what exactly is going on with the status, waiting, and background commands, I need you to do two tests:

- First, we need to see what exctly prompts some methods return ``0x0491`` reply. One possiblity is that it is still being transferred in the background. The other is that it is just a standard reply to any unexpected or non-implemented commands such as the ones for setting velocity parameters. To check that, you can try the following code:
```python
with Thorlabs.KinesisMotor("38293204") as stage:
    print(stage.instr.read())  # flush the input buffer
    print(stage.recv_comm())
```
If the commands are still being sent in the background, the second ``print`` will show them. Otherwise, it will time out.

- Second, we need to check how the stage status reflects its motion. For that, you can try calling ``get_status`` method at various times while the stage is moving or not and see if there's any difference. For example, you can run something like
```python
import time
with Thorlabs.KinesisMotor("38293204") as stage:
    print(stage.get_status())
    stage.move_to(0,channel=1)
    time.sleep(5)  # make sure that the stage is at 0
    stage.move_to(1370,channel=1)
    print(stage.get_status())
    time.sleep(0.1)
    print(stage.get_status()) # hopefully, the stage is moving now
    time.sleep(5)
    print(stage.get_status())
```

Please, let me know what are the test results.

Sincerely,

Alexey.

zac-castillo, 2023-02-08:
Dear Alexey,

- Result of the first test
   The first print statement returns an empty bytes literal: `b''`
   The second print statement returns the error, `ThorlabsBackendError: backend exception: 'read returned less data than expected' ('read returned less data than expected')` 
- Result of the second test
Here is the code I ran:
```python
import time
with Thorlabs.KinesisMotor("38293204") as stage:
    print(f'Status of stage upon connection: {stage.get_status()}\n')
    
    stage.move_to(0,channel=1)
    time.sleep(1)  
    print(f'Status as stage should be moving to zero: {stage.get_status()}\n')
    time.sleep(8) # The stage is definitely stopped at 0 after this wait
    print(f'Status as stage is stopped at zero and stationary: {stage.get_status()}')
    print(f'Status as stage is stopped at zero and stationary: {stage.get_status()}\n')
    
    stage.move_to(1370,channel=1)
    print(f'Status as stage should be moving to 1370 (immediate call): {stage.get_status()}')    
    time.sleep(0.1)
    print(f'Status as stage should be moving to 1370 (100 ms wait): {stage.get_status()}\n') # hopefully, the stage is moving now
    
    
    time.sleep(8) # The stage is definitely stopped at 1370
    print(f'Status as stage is stopped at 1370 and stationary: {stage.get_status()}')
    print(f'Status as stage is stopped at 1370 and stationary: {stage.get_status()}\n')
```
And the result:
```
Status of stage upon connection: ['enabled']

Status as stage should be moving to zero: ['moving_bk', 'enabled']

Status as stage is stopped at zero and stationary: ['moving_bk', 'enabled']
Status as stage is stopped at zero and stationary: ['enabled']

Status as stage should be moving to 1370 (immediate call): ['enabled']
Status as stage should be moving to 1370 (100 ms wait): ['moving_fw', 'enabled']

Status as stage is stopped at 1370 and stationary: ['moving_fw', 'enabled']
Status as stage is stopped at 1370 and stationary: ['moving_fw', 'enabled']
```
I find it interesting that two calls of `get_status()` at position 0 changed the result, but two calls at position 1370 did not.

Please let me know if you would like me to try any other tests. I should be able to try them in a more timely manner this time around.

-Zac

AlexShkarin, 2023-02-08:
Dear Zac,

That's still confusing. The first test suggests that there are no background commands, but the previous issues related to the wrong return commands seem to indicated that there are still some rogue ``0x0491`` messages being sent. It could, potentially, be that the motor sends them only when something special happens, e.g., when it starts or stops moving. Therefore, the statuses that you receive start lagging behind, first by one and then by two.

Could you try running exactly the same status-reading code as before, but with the added line ``print(stage.instr.read())`` in the end, to see if there are some unread messages in the queue?

Sincerely,

Alexey.

zac-castillo, 2023-02-08:
Hello Alexey,

The controller returns the following: 
`b'd\x04\x01\x00\x01P\x91\x04\x0e\x00\x81P\x01\x00Z\x05\x00\x00\x01\x00\x00\x00\x00\x04\x00\x80\x91\x04\x0e\x00\x81P\x01\x00Z\x05\x00\x00\x01\x00\x00\x00\x00\x04\x00\x80'`

Best,
-Zac

AlexShkarin, 2023-02-09:
Dear Zac,

Thank you for the tests!

I've updated the Kinesis code, [here](https://raw.githubusercontent.com/AlexShkarin/pyLabLib/bbe8c27b957cb5f41056505575371e9dc50cc88f/pylablib/devices/Thorlabs/kinesis.py) is the new version. As before, just use it to replace ``Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py`` in your Python folder.

Ideally, all of the issues should be now fixed:

- No methods should now malfunction because of stray ``0x0491`` replies
- Getting status should work reliably, including inside ``wait_for_move`` (i.e., it should show the stage status as moving immediately after the move was initiated)
- Similarly, ``get_position`` should also work correctly now
- I've re-implemented ``move_by`` for these stages with a combination of ``get_position`` and ``move_to`` (since they do not support this command natively), so now it should work. Keep in mind, though, that it does not check for boundaries, so you want to generally make sure that your destination is still between 0 and 170 degrees.
- I've added a couple of methods ``get_polctl_parameters`` and ``setup_polctl`` to query and set the polarization controller parameters (speed, homing position, and jog sizes). As far as I could see, these are the only adjustable parameters for these devices.
- I've also now implemented scaling for these stages, so if you now supply ``scale="stage"`` in the constructor, all relevant distances should be in degrees instead of steps. For example,
```python
with Thorlabs.KinesisMotor("38293204",scale="stage") as stage:
    stage.move_to(60)  # move to 60 degrees, not 60 steps
```
If you do not supply the ``scale`` parameter, then the behavior is the same as before.

Let me know if everything works, or if there some issues still left.

Sincerely,

Alexey.

zac-castillo, 2023-02-16:
Dear Alexey,

- I can step through a range of angles, wait until motion is stopped, and get the current position for both channels without issue.
- Getting the status appears to be working reliably, even when the stage is at 100% velocity
- `move_by()` works great
- 'get_polctl_parameters' and `setup_polctl ` work
- stage scaling works and is very convenient 

This is more than enough for my current purposes, thank you very much! I'll mark this as closed.

Sincerely,
Zac