AlexShkarin/pyLabLib #33: pylablib.devices.IMAQ.IMAQ.IMAQCamera: read_trigger method is not asserting when external trigger is pressed

jjcrouch, 2023-02-17:
Hi!

I am working on a script which grabs frames from a c-arm camera using a PCIe-1433 frame grabber. The frames must be grabbed when a pedal is pressed down; pressing and holding the pedal results in a repeated rising voltage edge at a consistent rate until the pedal is no longer pressed. In other words, the frame grabber must be synchronized with the camera capture. I also have a working LabView block diagram which provides me the correct trigger parameters, so I know my configure_trigger_in method is set up correctly.

The issue that I keep coming up against is that the read_trigger method is not asserting when I press down the external trigger pedal. I want to make 100% sure that I am correctly setting up my trigger line code, so I can turn my attention to the hardware itself if necessary.

Here is a script which first opens the frame grabber connected to the camera, then configures the input trigger, then continuously reads the trigger line for a rising edge. When there is a rising edge, it prints "asserted". I have also set up a timer which cancels the while loop after 30 seconds.

```
from pylablib.devices import IMAQ
import time

# identify frame grabber
cam = IMAQ.IMAQCamera('FPD::0')

# configure trigger in with corretc parameters
cam.configure_trigger_in('ext', trig_line=0, trig_pol='high', trig_action='buffer', timeout=1000.0)

# set 30 sec timer for while loop
timeout = time.time() + 30

# while loop continuously reads trigger line
while True:
    
    # checks trigger line value: 
        # if 0, then unasserted -> continue
        # if not 0, then asserted -> print "asserted" then continue
    while cam.read_trigger('ext', trig_line=0, trig_pol='high') == 0:
        
        # checks timer
        if time.time() > timeout:
            print('timeout')
            break
    
    else:
        print('asserted')
        
        # checks timer
        if time.time() > timeout:
            print('timeout')
            break
        
        continue
    
    break

cam.close()
```

A successful read_trigger line would print multiple "asserted" whenever the external trigger pedal is pressed. However, I get no "asserted" for the entire 30 sec timer when I am pressing the pedal.

Is there something wrong with my script's trigger setup?

AlexShkarin, 2023-02-18:
Hello!

As far as I can see, the code should work as intended, in the sense of repeatedly reading the current state of the trigger line and checking if it is asserted. There are several possible issues I can think of:

- Hardware connection issues, e.g., the pedal does not work, is not powered up, or is not connected to the frame grabber correctly.
- Some strange interaction between ``configure_trigger_in`` and ``read_trigger``. In principle, nothing like this seems to be mentioned in the documentation, but I've personally never used them simultaneously on the same line (see below).
- Short trigger pulse width. Keep in mind, that ``read_trigger`` reads the instantaneous state of the trigger line, so it can easily miss short trigger pulses. It looks like one call to this method takes about 50us, so if your pulses are significantly shorter than that, and their repetition rate is low, you might end up missing all of them. That is a relatively unlikely scenario, though, since I do not think trigger pulses shorter than 1us are very common.

There is also the question of what exactly you want to get from the trigger. If you want to make sure that the frame grabber reads and transfers a single frame every time the pedal is pressed, you do not necessarily have to use ``read_trigger`` method. All you need is to set up the external trigger (which you already do in your code) and start acquisition. The frame grabber hardware will then monitor the trigger line, and whenever the line get asserted, it will transfer a single frame. Hence, to test if the triggering works, you can try running the following code:
```python
from pylablib.devices import IMAQ
with IMAQ.IMAQCamera('FPD::0') as cam:
    cam.configure_trigger_in('ext', trig_line=0, trig_pol='high', trig_action='buffer', timeout=1000.0)
    cam.snap(timeout=10)  # read and return a single frame
```
If you run it and do not touch the pedal, ``cam.snap()`` should raise an ``IMAQTimeoutError`` after 10 seconds. Conversely, if you press the pedal and it sends a pulse, the frame will be acquired and the execution will be completed normally (as ``snap`` acquires only a single frame).

If, on the other hand, if you want to be able to check the trigger state in software to alter the script behavior (i.e., effectively use it as a universal digital input), you should use ``read_trigger`` instead of ``configure_trigger_in``. However, because of the pulse skipping issue I mentioned, I would recommend a non-pulsing kind of pedal, which simply asserts the line to the constant high value when pressed and de-asserts it when released.

Finally, you should remember that, unlike with standalone cameras, triggering the frame grabber only initiates frame transfer to RAM, but not necessarily camera frame acquisition. Therefore, it can often result in camera frames being skipped (if the camera frame rate is higher than the trigger rate), or in not all trigger pulses leading to a readout (if the camera frame rate is lower than the trigger rate). I think, in most cases what you want to do is to keep the frame grabber in the standard "free-running" mode (where it simply transfers all the frames generated by the camera) and instead trigger the camera directly. However, this is very hardware-dependent, so I can not recommend anything in particular.

Let me know, if that clarifies your issue. Also, if you already have a piece of LabView code which performs the task you want, you can send it to me (either here, or to pylablib@gmail.com), and I can try to check if it agrees with whatever the Python code does.

Sincerely,

Alexey.

jjcrouch, 2023-02-27:
I emailed pylablib@gmail.com with the following message, I just wanted to put it here to make sure you see it:

Hi Alexey,

Thank you so much for your prompt and detailed response! I have been unable to test the trigger test code you sent thus far, but I will be testing it early next week.

I am intrigued by the "free-running" mode that you mentioned in your response; I have a PCIe-1433 frame grabber and was wondering if this mode was something I could set up in python using the pylablib library?
It is crucial that we can get the frame grabber to transfer all of the images acquired by the camera (camera only acquires when trigger is pressed). This would solve the synchronization problem, as the frame grabber would only be saving the images that have been captured by the camera.

I have also attached my working LabView file [attached in the email sent to pylablib@gmail.com because Github doesn't accept the .vi file format]. It would be very helpful if you could let me know if it is possible to replicate its processes in Python?
As I mentioned above, I am most interested in saving all the frames acquired by the camera whenever I run the python program. I assume this would be achieved using a while loop that waits for and then saves each new frame in the camera buffer (as I am pretty sure this is roughly what the LabView file is doing).

Thank you so much for all your help so far!

Best regards,
James

AlexShkarin, 2023-03-02:
Dear James,

Sorry for the delayed reply!

By the "free-running" mode I mean just the default frame grabber mode, which it uses when you do not set up any external triggering (i.e., you do not need to do anything special to use it). In this mode, it simply tries to read out and transfer all the frames that the camera generates. In my experience with NI frame grabbers, when you combine it with externally triggering the camera, it does pretty much what you want it to do: acquiring frames only on the external trigger pulses (camera takes care of that) and transferring them to RAM.

The code for continuous readout would look something like this:
```python
from pylablib.devices import IMAQ
with IMAQ.IMAQCamera('FPD::0') as cam:
    frames=[]
    max_frames=100
    cam.set_roi(0,1024,0,1024)  # set up the frame size
    # configure the trigger; I don't think you need this line if you trigger the camera directly, so it's commented out
    # cam.configure_trigger_in('ext', trig_line=0, trig_pol='high', trig_action='buffer', timeout=1000.0)
    cam.setup_acquisition(nframes=100)  # setup acquisition and specify the internal buffer size to store transferred frames
    # note that this does not mean that the grabber will only acquire 100 frames; rather, it means that it can store
    # up to 100 transferred frames which have not yet been read by the script
    # you typically want the buffer size to be ~1s worth of data, e.g., 1000 frames for 1kFPS rate
    cam.start_acquisition()  # starting from this point the frame grabbers starts transferring frames
    while len(frames)<max_frames:  # wait until we've acquired 100 frames; you can put other stop conditions
        rng=cam.get_new_images_range()  # check if new frames are avilable
        if rng is not None:  # got new frames
            # read new frames and append to the list
            # frames from rng are marked as read, so they will not be included in the next get_new_images_range call
            frames+=cam.read_multiple_images(rng)
        else:
            time.sleep(1E-3)  # sleep for 1ms between checks to reduce CPU load
    cam.stop_acquisition()
```
As you correctly noted, for the most part it is a while loop that periodically checks if there are any new frames and appends them to a list (you can also, e.g., directly save them to the drive instead). As far as I could see, it more or less corresponds to the LabView code you sent.

Let me know if that code works for you, or if you have any questions about it.

Sincerely,

Alexey.

jjcrouch, 2023-03-31:
Hi Alexey,

Hope you are well.

I was able to get the following code to transfer frames, but only at lower frame rates. At higher frame rates, frames are skipped:

`# creates list of frames
    frames = []
    
    # creates frame counter
    counter = 0
    
    # opens and closes camera automatically
    with IMAQ.IMAQCamera('FPD::0') as cam:
    
        # configures external trigger
        cam.configure_trigger_in('ext', trig_line=0, trig_pol='high', 
                                 trig_action='buffer', timeout=1000.0)
        
        while True:
            
            try:
                # waits for trigger and snaps frame
                frame = cam.snap(timeout=5.0)

                # saves frame to frames list
                frames.append(frame)
                
                # puts the frame into the queue
                queue.put(frame)
                
                # adds to frame counter
                counter += 1
                
            except:
                break
        
        # call save frames function
        save_frames(frames, counter)`

I wanted to make sure that this skipping frames issue wasn't due to my equipment, so I used your continuous acquisition code you provided in your response and it successfully transferred all frames. Sadly, I cannot use this method, because I need to only capture frames containing image data and this method captures blank frames as well.

The conclusion I've come to is that the snap method within the while loop is taking longer that it takes the camera to capture the next frame. This results in many frames being skipped. 

So, I am wondering if there is a way to reduce the time it takes for the snap method to complete? Or if there is any alternative to the snap method that is able to only transfer frames containing image data?

Thanks again for all your help with this!

Best regards,
James

AlexShkarin, 2023-04-08:
Hi James,

You are right, the snap method is generally much slower, because it starts and stops acquisition for every frame, and that takes some time. If you want to avoid it, you'll need to just use the acquisition loop I've posted before.

I'm a bit surprised to hear about these blank frames. Can you described them more? Are these simply all-zero frames? Do you also see them when using NI MAX to look at the frame grabber acquisition? Are they described in your camera documentation?

In any case, I would personally try to still keep the acquisition loop as before at address blank frames separately, say, by just ignoring them if they are easily identifiable. For example, if they are all zeros, you can modify the loop like
```python
 while len(frames)<max_frames:
    rng=cam.get_new_images_range()
    if rng is not None:
        new_frames=cam.read_multiple_images(rng)
        new_frames=[f for f in new_frames if np.any(f!=0)]  # skip frames which are all zeros
        frames+=new_frames
    else:
        time.sleep(1E-3)
```

Sincerely,

Alexey.