AlexShkarin/pyLabLib #34: Can't list devices because FT232DeviceBackend won't load correctly

liam-clink, 2023-02-23:
I have pyft232 installed automatically as a dependency of pylablib, and when I'm just trying to list connected Thorlabs Kinesis devices, I get this error.

```python
Traceback (most recent call last):
  File "/home/liam/Documents/GitHub/shg-polarization-scan/main.py", line 118, in <module>
    print(pylablib.devices.Thorlabs.list_kinesis_devices())
  File "/home/liam/user_python/lib/python3.10/site-packages/pylablib/devices/Thorlabs/kinesis.py", line 26, in list_kinesis_devices
    return KinesisDevice.list_devices(filter_ids=filter_ids)
  File "/home/liam/user_python/lib/python3.10/site-packages/pylablib/devices/Thorlabs/kinesis.py", line 101, in list_devices
    dids=comm_backend.FT232DeviceBackend.list_resources(desc=True)
AttributeError: module 'pylablib.core.devio.comm_backend' has no attribute 'FT232DeviceBackend'. Did you mean: 'PyUSBDeviceBackend'?
```

AlexShkarin, 2023-02-24:
It looks like you're using it on a Linux system, which has slighly different conventions. Instead of ``pyft232``, it uses just plain ``pyserial`` interface, so ``list_kinesis_devices`` does not work there. To make it work, first you need to figure out which dev path (e.g., ``/dev/ttyUSB0``) corresponds to the device by, for example, plugging and unplugging it and checking which path appears and disappears. Then you need to supply this path to the class to open it. The current pylablib implementation has problems with accepting ``pyserial`` paths right aways, so the code will be a bit verbose:
```python
conn = {"port":"/dev/ttyUSB0","baudrate":115200,"rtscts":True}  # intead of ttyUSB0 use the correct path
dev = Thorlabs.KinesisMotor(("serial",conn))  # can also be, e.g., KinesisPiezoMotor, depending on the device kind
```
Let me know if that solves the issue.

liam-clink, 2023-03-03:
So it isn't possible to use list_kinesis_devices() at all when not on Windows?

AlexShkarin, 2023-03-04:
That's right, since it uses ``pyft232``, this function can only work on Windows.

Nevertheless, if you manage to identify the dev path for your device, you still can use pylablib to control devices on other operating systems (at least, on Linux).