AlexShkarin/pyLabLib #35: Jupyter note

Buantum, 2023-03-08:


AlexShkarin, 2023-03-15:
Hello!

As far as I can tell, Ultra 888 could be an SDK2 camera, not SDK3 (these both come from Andor, but the interfaces are completely different). So could you try running the following:
```python
import pylablilb as pll
pll.par["devices/dlls/andor_sdk2"]="D:/Andor SOLIS"
from pylablilb.devices import Andor
print(Andor.get_cameras_number_SDK2())
```
and see what number it prints?

Alexey.

Buantum, 2023-03-16:
it prints 1, which means it is successfully connected？
[image]

AlexShkarin, 2023-03-16:
Great! To make sure that the connection work, you can try to execute something like
```python
import pylablilb as pll
pll.par["devices/dlls/andor_sdk2"]="D:/Andor SOLIS"
from pylablilb.devices import Andor
with Andor.AndorSDK2Camera() as cam:
    print(cam.get_full_info())
    print(cam.snap())
```

Regarding the related issue (AlexShkarin/pyLabLib-cam-control#1): since your Andor SOLIS is installed in a non-standard folder, the software can not find the necessary libraries (that is also why you need to specify ``pll.par["devices/dlls/andor_sdk2"]`` in the Python code). To fix that, you can find ``settings.cfg`` file inside ``cam-control/cam-control`` folder and add a line
```
dlls/andor_sdk2 D:/Andor SOLIS
```
Afterwards, the software should find the camera. Just in case, you can also run ``detect.exe`` before ``control.exe``, so that it would re-scan for all available cameras.

Let me know if that works!

Alexey.

Buantum, 2023-03-16:
it works successfully. and I'd want to know whether there are some customize-mathematical operations in cam-control files?

AlexShkarin, 2023-03-19:
I'm glad it works!

Regarding the cam-control: do you mean real-time operation? There are some [basic algorithms](https://pylablib-cam-control.readthedocs.io/en/latest/advanced.html#filters) built-in, such spatial band-pass filtering, background subtraction, and rolling average/median/min/max/standard deviation. In addition, you can write your own [filters](https://pylablib-cam-control.readthedocs.io/en/latest/expanding.html#custom-filters) in Python.

However, I would recommend using these only to do some basic real-time image manipulation to quickly make sense of the camera data during the experiment. In general, I would suggest saving the raw camera data and process it in the software of your choice (e.g., Python, Matlab, ImageJ). This way, you can always try a different analysis scheme without having to re-take new data.

Alexey.

Buantum, 2023-03-20:
Thank you for your enthusiastic answer！
In our lab，we want to manipulation the real-time image with some easy operations  to witness some fast process in real-times and then determine the parameters of next step. In our previous setting in Labview, the operations are quickly completed and so ignore the time. 
Thanks for your reply, and I will learn the filter.

Buantum, 2023-03-20:
I have viewed base.py in filters. It looks that if I want to manipulate image autonomously，I just put my code into process_frame() in each class？

AlexShkarin, 2023-03-20:
No, you don't need to alter the existing classes. You can just define a new class in a new file located in the same ``plugins/filters`` folder. To get a better idea, you can check out ``template.py`` in the same folder as an example.

It looks like a single-frame filter (defilend in ``TemplateSingleFrameFilter``) might be enough for you. This kind of filter simply takes a single image (a new frame) and returns a modified image which is displayed in GUI. To define it from the template, you can do the following:

- Make a copy of the ``template.py`` file in the same folder with an arbitrary name (e.g., ``myfilter.py``)
- Uncomment ``_class_name="template"`` line inside the ``TemplateSingleFrameFilter`` class definition. It is also a good idea to rename it to something else, e.g., ``_class_name="myfilter"``; otherwise, there might be conflicts when two filters define the same name
- Optionally, you can also change ``_class_caption`` and ``_class_description`` strings to better present your filter in the software.
- Finally, most importantly, you need to alter the ``process_frame`` method to do the image manipulation you want.

I hope that helps!

Alexey.

Buantum, 2023-03-21:
Thanks for reply and I have completed it！another question， in our lab，the camera is controled by sequence but not the botton in GUI. I want to know whether there are some method to involve pylablib-cam？
thanks！maybe I have too many questions to trouble you！

----------

该邮件从移动设备发送

-----------------------------------

AlexShkarin, 2023-03-21:
I'm not sure exactly what you mean... Could this be realted to external / internal camera triggering? You can change these in the ``Advanced`` camera settings and see if it has the effect you want.

Buantum, 2023-03-22:
Let me describe exactly....in our lab, we capture the images through sequence but not in the andor soli which we only use to set parameters. For some reasons, we need to display image as fast as possible and so I find the pylablib. In pylablib-cam, if I want to capture images, I should operate in GUI of pylablib-cam(however，setting parameters in pylablib-cam is fine for us). Now， what we need is there is a sequence to let camera capture which is not on the GUI and the images data display on pylablib-cam GUI after the mathematical operation mentioned above. thank you！