AlexShkarin/pyLabLib #36: SiliconSoftware compatibility broken with MicroEnable V

matpelb, 2023-03-20:
Some attribute names have changed in the SDK from MicroEnable IV to MicroEnable V, which leads to crashes when instantiating a Camera object relying on MicroEnable V:

- CAMERA_LINK_CAMTYP is named CAMERA_LINK_CAMTYPE in MicroEnable V SDK
- FG_GLOBAL_ACCESS is not defined for MicroEnable  V applets, only in MicroEnable IV

By surrounding the few calls to these attributes with try/catch block (using CAMERA_LINK_CAMTYPE when CAMERA_LINK_CAMTYP fails) I have managed to make my application work with MicroEnable V framegrabber as it did with MicroEnable IV

AlexShkarin, 2023-03-20:
Thanks for letting me know! It will be fixed in the next release.

Jasondio, 2023-05-16:
Hi, I recently had a problem using this software, and my frame grabber is also MicroEnable V. After opening the detect program, it shows that the camera is skipped and the camera is not opened by other software, my camera model is PhotonFocus MV-D1024E. the solution you mentioned I also tried to use, but No problem solved.   I look forward to your answer, your answer will be of great help to me, thank you. @matpelb 

The following is the message printed by the detect program

Searching for PhotonFocus cameras (might take several minutes)
Checking potential PFRemote interfaces Silicon Software GmbH/board0_port0, Silicon Software GmbH/board0_port1

Checking interface Silicon Software GmbH/board0_port0 ... discovered camera MV1-D1024E-160
Found 1 PhotonFocus camera
Checking PhotonFocus camera idx=0
	Port board0_port0,   vendor Silicon Software GmbH,   model MV1-D1024E-160
Skipping the camera

matpelb, 2023-05-22:
Hi @Jasondio 
If the camera is not detected/opened by both the PFremote interface and MicroDisplay  the problem is most probably not related to pylablib. 
Is that what you mean by "the camera is not opened by other software" ?