AlexShkarin/pyLabLib #42: Incorrect camera attribute limits

bhathawayML, 2023-06-23:
Hello,

While using pylablib.devices.IMAQdx, I've noticed that the camera attribute limits for exposure are being read as [0.5, 250] ms rather than the actual camera limits which are [0.1, 2000] ms (determined by using the manufacturer's software). This particular camera is from AmScope and I have found online that they have some incompatibilities with IMAQdx. Is it possible to set these parameters manually? Even when I delete the icd camera file IMAQdxOpenCamera recreates the same one, so I'm assuming these values are hard-coded somewhere...

I know this is more of a particular problem with the IMAQ library and not necessarily an issue in pylablib, but thank you in advance for any help :)

AlexShkarin, 2023-06-26:
To override the truncation in the pylablib code you can supply ``truncate=False`` to the setting function. That is, instead of writing
```python
cam.cav["CameraAttributes/Exposure/Value"]=500
```
or
```python
cam.set_value("CameraAttributes/Exposure/Value",500)
```
(both are equivalent), you can write
```python
cam.set_value("CameraAttributes/Exposure/Value",500,truncate=False)
```
This migh help if IMAQdx reports incorrect parameter limits (using ``IMAQdxGetAttributeMinimum`` and ``IMAQdxGetAttributeMaximum``), but allows values outside these limits in ``IMAQdxSetAttribute``. However, if setting this parameter causes an error directly from IMAQdx (i.e., ``IMAQdxLibError``), there's, unfortunately, nothing you can do. As far as I can tell, IMAQdx does not allow overriding its internal limits.

bhathawayML, 2023-06-26:
Unfortunately, I got the IMAQdxLibError error, but thank you anyways for the insight!

closed 2023-06-26 by bhathawayML