AlexShkarin/pyLabLib #43: BUG: Cannot create Thorlabs.KinesisMotor from a serial number

kadykov, 2023-06-26:
Cannot connect to a Thorlabs KST101 motor controller. `pylablib.devices.Thorlabs.list_kinesis_devices()` detects a correct serial number, but creation an instance of `pylablib.devices.Thorlabs.KinesisMotor(sernumber)` crashes with `ThorlabsBackendError: backend exception: 'read returned less data than expected'`.

Minimal example to reproduce:
```python
from pylablib.devices import Thorlabs

devices = Thorlabs.list_kinesis_devices()
sernumber = devices[0][0]
print(sernumber)

Thorlabs.KinesisMotor(sernumber)
```
<details>
<summary>Output</summary>

```
26000280
---------------------------------------------------------------------------
ThorlabsBackendError                      Traceback (most recent call last)
[c:\Users\Username\Documents\prog\packagename\packagename\temp\thorlabs-stage-dev.py](file:///C:/Users/Username/Documents/prog/packagename/packagename/temp/thorlabs-stage-dev.py) in line 8
      [5](file:///c%3A/Users/Username/Documents/prog/packagename/packagename/temp/thorlabs-stage-dev.py?line=4) sernumber = devices[0][0]
      [6](file:///c%3A/Users/Username/Documents/prog/packagename/packagename/temp/thorlabs-stage-dev.py?line=5) print(sernumber)
----> [8](file:///c%3A/Users/Username/Documents/prog/packagename/packagename/temp/thorlabs-stage-dev.py?line=7) Thorlabs.KinesisMotor(sernumber)

File [c:\Users\Username\mambaforge\envs\envname\Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py:1031](file:///C:/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py:1031), in KinesisMotor.__init__(self, conn, scale, default_channel, is_rack_system)
   [1029](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1028) self._add_settings_variable("limit_switch_parameters",self.get_limit_switch_parameters,self.setup_limit_switch)
   [1030](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1029) with self._close_on_error():
-> [1031](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1030)     self._stage=self._get_stage(scale)
   [1032](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1031)     self._scale,self._scale_units=self._calculate_scale(scale)
   [1033](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1032) if self.get_device_info().model_no in ["KDC101","KST101","KBD101"]:

File [c:\Users\Username\mambaforge\envs\envname\Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py:1048](file:///C:/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py:1048), in KinesisMotor._get_stage(self, scale)
   [1047](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1046) def _get_stage(self, scale):
-> [1048](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1047)     model=self.get_device_info().model_no
   [1049](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1048)     if scale=="stage":
   [1050](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=1049)         return self._autodetect_stage(model)

File [c:\Users\Username\mambaforge\envs\envname\Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py:201](file:///C:/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py:201), in BasicKinesisDevice.get_device_info(self, dest)
    [195](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=194) def get_device_info(self, dest="host"):
    [196](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=195)     """
    [197](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=196)     Get device info.
    [198](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/devices/Thorlabs/kinesis.py?line=197) 
...
--> [922](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/core/devio/comm_backend.py?line=921)         raise self.Error("read returned less data than expected")
    [923](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/core/devio/comm_backend.py?line=922) self.cooldown("read")
    [924](file:///c%3A/Users/Username/mambaforge/envs/envname/Lib/site-packages/pylablib/core/devio/comm_backend.py?line=923) self._log("read",result)

ThorlabsBackendError: backend exception: 'read returned less data than expected' ('read returned less data than expected')
```
</details>

Thorlabs KST101 connected to USB3 port via Micro Type B cable
OS: Windows 10
Python: 3.10.12, 3.11.3
Pylablib: 1.4.1
pyft232: 0.12 

<details>
<summary>Full environment</summary>

```
# Name                    Version                   Build  Channel
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
asciitree                 0.3.3                      py_2    conda-forge
astropy                   5.3             py311h59ca53f_0    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
async_generator           1.10                       py_0    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
aws-c-auth                0.6.28               habb20d3_5    conda-forge
aws-c-cal                 0.5.27               hc655790_0    conda-forge
[... 304 lines cut ...]
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
urllib3                   2.0.3              pyhd8ed1ab_0    conda-forge
vc                        14.3                hb25d44b_16    conda-forge
vc14_runtime              14.36.32532         hfdfe4a8_16    conda-forge
vs2015_runtime            14.36.32532         h05e6639_16    conda-forge
wcwidth                   0.2.6              pyhd8ed1ab_0    conda-forge
webencodings              0.5.1                      py_1    conda-forge
wheel                     0.40.0             pyhd8ed1ab_0    conda-forge
widgetsnbextension        4.0.7              pyhd8ed1ab_0    conda-forge
win_inet_pton             1.1.0              pyhd8ed1ab_6    conda-forge
xarray                    2023.5.0           pyhd8ed1ab_0    conda-forge
xorg-libxau               1.0.11               hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xyzservices               2023.5.0           pyhd8ed1ab_1    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yawrap                    0.4.10                   pypi_0    pypi
zarr                      2.15.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zict                      3.0.0              pyhd8ed1ab_0    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.2                h12be248_6    conda-forge
```
</details>

AlexShkarin, 2023-06-26:
Unfortunately, the traceback got truncated in the middle, but it looks like the device is simply not replying to any commands (since this is the first time the code expects a reply).

To double-check, could you run the following code
```python
dev=Thorlabs.BasicKinesisDevice(sernumber)
dev.blink()
dev.get_device_info()
```
and check whether
- it passes or raises an error
- the controller reacts in any way (e.g., turns on its screen light)?

Also, just to check that it's not a hardware issue: can you connect to the same device using the Thorlabs Kinesis software? And have you tried pylablib for other Kinesis devices, connected to the same PC, if you have any?

kadykov, 2023-06-27:
You are right, it was just a problem with the controller itself. A proper reboot solved the problem.
It seems that a quick on-off power cycle does not actually reboot the device, it is necessary to wait at least 5 seconds before switching it on again.

Thank you very much for your help.

closed 2023-06-27 by kadykov