AlexShkarin/pyLabLib #44: Is it possible to control gain of the Thorlabs camera?

khoroshyy, 2023-06-28:
Hi 
I am using pylablib with Thorlabs Kiralux camera cs135mu but I cannot figure out how to control gain of the camera. Is it implemented?
Thanks.
Petro.

AlexShkarin, 2023-07-01:
Hello!

Unfortunately, the camera class does not currently implement the gain control. Nevertheless, the required method wrappers are already defined in the backend. Therefore, you can simply use the following functions to control the gain:
```python
from pylablib.devices import Thorlabs

def get_gain(camera):
    idx=Thorlabs.TLCamera.lib.tl_camera_get_gain(camera.handle)
    return Thorlabs.TLCamera.lib.tl_camera_convert_gain_to_decibels(camera.handle,idx)

def get_gain_range(camera):
    rng=Thorlabs.TLCamera.lib.tl_camera_get_gain_range(camera.handle)
    return tuple(Thorlabs.TLCamera.lib.tl_camera_convert_gain_to_decibels(camera.handle,g) for g in rng)

def set_gain(camera, gain):
    idx=Thorlabs.TLCamera.lib.tl_camera_convert_decibels_to_gain(camera.handle,gain)
    Thorlabs.TLCamera.lib.tl_camera_set_gain(camera.handle,idx)
```

Keep in mind that trying to set the gain outside of the range would cause an error.

Sincerely,

Alexey.

khoroshyy, 2023-10-07:
Thanks a lot. I will try it.