AlexShkarin/pyLabLib #49: Thorlabs Kinesis Motor "Device Not Opened" Error

bhathawayML, 2023-07-26:
Hello! 

I am trying to communicate with a Thorlabs stage (HDR50) with the BSC201 Stepper Motor Controller, but I keep getting the following error when calling ```pylablib.devices.Thorlabs.KinesisMotor("40330594")```:

```
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2023.1\plugins\python-ce\helpers\pydev\_pydevd_bundle\pydevd_exec2.py", line 3, in Exec
    exec(exp, global_vars, local_vars)
  File "<input>", line 1, in <module>
  File "C:\Users\lab\Documents\GitLab\tetons\venv\lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 999, in __init__
    super().__init__(conn)
  File "C:\Users\lab\Documents\GitLab\tetons\venv\lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 246, in __init__
    super().__init__(conn,timeout=timeout,is_rack_system=is_rack_system,default_axis=default_channel)
  File "C:\Users\lab\Documents\GitLab\tetons\venv\lib\site-packages\pylablib\devices\interface\stage.py", line 34, in __init__
    super().__init__(*args,**kwargs)
  File "C:\Users\lab\Documents\GitLab\tetons\venv\lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 45, in __init__
    instr=comm_backend.new_backend(conn,backend=("auto","ft232"),term_write=b"",term_read=b"",timeout=timeout,
  File "C:\Users\lab\Documents\GitLab\tetons\venv\lib\site-packages\pylablib\core\devio\comm_backend.py", line 1569, in new_backend
    return backend(conn,**kwargs)
  File "C:\Users\lab\Documents\GitLab\tetons\venv\lib\site-packages\pylablib\core\devio\comm_backend.py", line 813, in __init__
    raise self.Error(e) from e
pylablib.devices.Thorlabs.base.ThorlabsBackendError: backend exception: D2XXException(3) ('Device Not Opened')
```

I can use ```pylablib.devices.Thorlabs.list_kinesis_devices()``` or ```print(comm_backend.list_backend_resources("ft232"))``` to show the device is detected:

_[('40330594', 'APT Stepper Motor Controller')]_

...and I can control the device in the Kinesis software ok.

Things I've tried with no luck:
1. Using ```pylablib.devices.Thorlabs.BasicKinesisMotor``` (encountered same error)
2. Using Python 3.6 and 3.10 (same issue in both)
3. Using pylablib v1.4.1 and v1.4.0 (same issue in both)
4. Reinstalling pylablib
5. Power cycling, checking physical connections, etc.
6. Using thorlabs_apt module which _does_ seem to work, but I would prefer using pylablib :)

Thanks in advance!

**Specs**: Windows 10 64bit, PyCharm 2023.1.4, python 3.10, pylablib v1.4.1

AlexShkarin, 2023-10-08:
Hello!

Thank you for the very detailed report, and sorry for the later reply!

The only possibility I can see is that the device is opened in some other software (APT, Kinesis, or another Python instance). This usually does not interfere with the device discovery, but it does prevent actually connecting to the device. To check this, I would suggest restarting the PC and only start one Python interpreter instance with your code. If that still doesn't work (i.e., it shows the same exact error as before), you can also try using ``pyft232`` directly:
```python
import ft232
inst=ft232.Ft232("40330594",115200,rtscts=False)  # assuming '40330594' is your device serial number
inst.close()
```
(again, make sure that the device is not being used in any other software). If that still raises an error, then the only explanation I can come up with is some problem with ``pyft232`` package or the corresponding drivers.

bhathawayML, 2023-10-13:
Thanks Alex!

Turns out my stepper motor was broken and need to be returned to Thorlabs for repairs. Our new one works fine with your code :)

Thanks again for the help.

closed 2023-10-13 by bhathawayML