AlexShkarin/pyLabLib #54: Exposure time in Basler camera broken

bmoneke, 2023-10-19:
https://github.com/AlexShkarin/pyLabLib/blob/c3521c8ad2fa6064b3f8a117576ed3053b7fa69f/pylablib/devices/Basler/pylon.py#L452

Getting setting exposure time does not work.
My camera has a "ExposureTime" attribute but not "ExposureTimeAbs", as the get all config shows.

If I remove the "Abs", I can get / set the exposure time.

Cameras are Basler Dart daA1280-54um and daA2500-14um

bmoneke, 2023-10-21:
By the way, Basler offers their own wrapper, which might reduce your maintanence effort:

https://github.com/basler/pypylon

AlexShkarin, 2023-10-22:
Thanks for pointing out the issue! I will fix it in the next release.

bmoneke, 2023-10-25:
I also noticed, that the methods `get_roi` and `set_roi` of `BaslerPylonCamera` do not match the superclass IROICamera definition.
The Basler's methods only handle the roi without binning (4 parameters), while the IROICamera handles roi and binning (6 parameters).