AlexShkarin/pyLabLib #55: Newport Module not detecting picomotors

jayshah15028, 2023-10-20:
I have the Newport picomotor controller 8742 connected to the computer using a USB port. I am trying to run this simple 2 line code, but its output is 0, instead of 2 (I have 2 picomotors connected to the controller).
```python

from pylablib.devices import Newport
print(Newport.get_usb_devices_number_picomotor())

AlexShkarin, 2023-10-22:
The library identifies the controllers based on their USB IDs, so it could be that these are different for your controller models. Can you run
```python
from pylablib.core.devio import comm_backend
print(comm_backend.PyUSBDeviceBackend.list_resources(idVendor=0x104d))
```
and post the output? When doing that, you should make sure that no other software is using the Picomotor controllers when you to connect to them, as that may cause conflicts.

jayshah15028, 2023-10-23:
Thanks for the response. The output of the code you gave is:

> []

It did not list any resources.

AlexShkarin, 2023-10-29:
Thanks for the info! So, it looks like there are no devices with the expected vendor ID 0x104D, which typically corresponds to Newport. Can you control the motors using the native Newport software? And can you find it in the device manager and see what their VID and PID are? In Windows those are usually in the ``Properties -> Details -> Hardware IDs``

Zhenye-eng, 2024-07-10:
Hi, Are you still interested in this? I encountered the same issue. The problem is caused by the incompatible driver on 8742. The default one is WINUSB(v5.0.7.0) which will not work with pyusb. You can change driver to libusbK using Zadig. I have a similar tuition changing driver on usb devices: https://learn.adafruit.com/circuitpython-on-any-computer-with-ft232h/windows. Also note that once you changed the driver, the New Focus Picomotor Application will fail.