AlexShkarin/pyLabLib #56: IMAQCamera.setup_acquisition() costs very long execuation time

jianrong2, 2023-10-22:
Hi,

Thanks for your contributions to the awsome project. I find it very useful to acquire high throughoutput video via the IMAQ wrapper.

The library works well though I found  a minor issue when initilizing the camera interface. It take much longer time than the native application such as the NI-MAX.

With the help of breakpoints, I located the line `self._max_nbuff=self._find_max_nbuff()` of the method ` def setup_acquisition(self, mode="sequence", nframes=100)` in [IMAQ.py](https://github.com/AlexShkarin/pyLabLib/blob/main/pylablib/devices/IMAQ/IMAQ.py), which might cause the issue. With hard-coded  `self._max_nbuff=5000` instead, the script runs much more smooth.

Notes: My setup is NI PCIe-1433 frame grabber combined with a GL2048R InGaAs line camera. And I use `cam0.setup_acquisition(mode="sequence", nframes=100)` to initilize the camera.

AlexShkarin, 2023-10-22:
Hi,

Thank you for the kind words, and for pointing out this issue!

Yes, it is a known problem with IMAQ that allocating buffers takes super-linear (almost exponentail at some point) time, and it can take a many seconds for more than ~30-50k frames. Since the code has to do it to find the maximal available bumber of buffers, that causes the slowdown.

I agree, this is not handled in the best possible way right now, and it should, probably, be a constructor parameters. If you want to avoid changing the pylablib source files (since it makes the software not transferable between different PCs), you can also call ``cam0._max_nbuff=5000`` right after opening the camera and before the first call to ``setup_acquisition`` or ``start_acquisition``.

jianrong2, 2023-10-30:
Thanks for your reply. I tried `cam0._max_nbuff=5000` instead of modifying the library source file,  and found it worked perfectly 😄.

closed 2023-10-30 by jianrong2