AlexShkarin/pyLabLib #57: without“DETECTOR.INI”

Buantum, 2023-10-22:
I try to load dll files to initialize Python class with code
[image]
However, it returns 20002 which means failed. 
I find that there is no “DETECTOR.INI” files in Andor SDK files which is necessary for initialize. So I learn the code of pylablib which also needs ini_path for class AndorSDK2camera. but why I run pylablib successfully eventhough there is no “DETECTOR.INI”？

AlexShkarin, 2023-10-22:
Hi!

Acutally, according to the documentation 20002 is the "success" error code (which, I agree, is unconventional):
[image]
Regarding ``DETECTOR.INI``: as far as I know, it is only requires for some specific cameras, mostly older models. Hence, by default ``pylablilb`` just supplies an empty string. According to the documentation, if the path is indeed required, and it's not supplied, the ``Initialize`` function will return ``DRV_INIERROR`` error (20070). However, I've never worked with such cameras, so I can't say what the actual behaviour would be.

Buantum, 2023-10-23:
thanks for your reply！
I have found such an conventional thing today which troubles me with “error”