AlexShkarin/pyLabLib #66: Andor Timeout Error

lrennaIOS, 2023-11-22:
Hello,

I have an Andor DB401. I am able to connect to the camera and get different status information. I can also open and close the shutter, etc. However, it times out whenever I try to get an image (with snap or grab).
---

cam = Andor.AndorSDK2Camera() # connect to Andor SDK2 camera (e.g., iXon)
print(cam.get_full_info())
cam.set_exposure(0.1) # set exposure to 100ms
frame = cam.snap() # grab a single frame
cam.close() # close the connection

{'image_indexing': 'rcb', 'frame_format': 'list', 'frame_info_format': 'namedtuple', 'frame_info_period': 1, 'roi': (0, 1024, 0, 127, 1, 1), 'temperature': -76, 'cooler': False, 'channel': 0, 'oamp': 0, 'hsspeed': 0, 'preamp': 0, 'vsspeed': 0, 'shutter': ('closed', 0, 0, 0), 'fan_mode': None, 'trigger_mode': 'int', 'acq_parameters/accum': (1, 0), 'acq_parameters/kinetic': (1, 0.0, 1, 0, 0), 'acq_parameters/fast_kinetic': (1, 0.0), 'acq_parameters/cont': 0, 'acq_mode': 'cont', 'frame_transfer': None, 'read_parameters/single_track': (0, 1), 'read_parameters/multi_track': (1, 1, 0), 'read_parameters/random_track': [(0, 1)], 'read_parameters/image': (0, 1024, 0, 127, 1, 1), 'read_mode': 'image', 'exposure': 0.0010000000474974513, 'frame_period': 0.14000000059604645, 'buffer_size': 0, 'acquired_frames': 0, 'acquisition_in_progress': False, 'frames_status': TFramesStatus(acquired=0, unread=0, skipped=0, buffer_size=0), 'data_dimensions': (127, 1024), 'frame_timings': TAcqTimings(exposure=0.0010000000474974513, frame_period=0.14000000059604645), 'roi_limits': (TAxisROILimit(min=2, max=1024, pstep=1, sstep=1, maxbin=32), TAxisROILimit(min=2, max=127, pstep=1, sstep=1, maxbin=32)), 'temperature_monitor': -33.23921585083008, 'temperature_status': 'not_reached', 'amp_mode': TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Conventional', hsspeed=0, hsspeed_MHz=1.0, preamp=0, preamp_gain=1.0), 'acq_status': 'idle', 'cycle_timings': TCycleTimings(exposure=0.0010000000474974513, accum_cycle_time=0.14000000059604645, kinetic_cycle_time=0.14000000059604645), 'readout_time': 9.999999747378752e-06, 'cls': 'AndorSDK2Camera', 'conn': (0, ''), 'detector_size': (1024, 127), 'frame_info_fields': ['frame_index'], 'device_info': TDeviceInfo(controller_model='CCI-010', head_model='DV401', serial_number=0), 'pixel_size': (2.6e-05, 2.6e-05)}

AndorTimeoutError                         Traceback (most recent call last)
<ipython-input-268-7a6dc76e25c5> in <module>
      2 print(cam.get_full_info())
      3 cam.set_exposure(0.1) # set exposure to 100ms
----> 4 frame = cam.snap() # grab a single frame
      5 cam.close() # close the connection

~\anaconda3\lib\site-packages\pylablib\devices\interface\camera.py in snap(self, timeout, return_info)
    680     def snap(self, timeout=5., return_info=False):
    681         """Snap a single frame"""
--> 682         res=self.grab(frame_timeout=timeout,return_info=return_info)
    683         if return_info:
    684             return _first_frame(*res,chunks=self.get_frame_format()=="chunks")

~\anaconda3\lib\site-packages\pylablib\devices\interface\camera.py in grab(self, nframes, frame_timeout, missing_frame, return_info, buff_size)
    666         try:
    667             while nacq<nframes:
--> 668                 self.wait_for_frame(timeout=frame_timeout)
    669                 if return_info:
    670                     new_frames,new_info,rng=self.read_multiple_images(missing_frame=missing_frame,return_info=True,return_rng=True)

~\anaconda3\lib\site-packages\pylablib\core\devio\interface.py in wrapped(*args, **kwargs)
    664         def wrapped(*args, **kwargs):
    665             all_args=parse_args(args,kwargs)
--> 666             res=func(**all_args)
    667             return parse_reply(res,args,kwargs)
    668         def wrapped_wip(*args, **kwargs):

~\anaconda3\lib\site-packages\pylablib\devices\interface\camera.py in wait_for_frame(self, since, nframes, timeout, error_on_stopped)
    260                 to=fto if to is None else min(to,fto)
    261             if to is not None and to<=0:
--> 262                 raise self.TimeoutError
    263             self._wait_for_next_frame(timeout=to,idx=acquired_frames)
    264         self._frame_counter.wait_done()

AndorTimeoutError:

AlexShkarin, 2023-11-28:
There doesn't seem to be any obvious reason for why it would time out. Does it work with the native software (i.e., Andor Solis)? 

Also, to further debug the issue, could you try the following code:

```python
import time

cam = Andor.AndorSDK2Camera()
print(cam.get_full_info())
time.sleep(10.)
print(cam.get_full_info())
```
This will show if any properties changed during the acquisition,

lrennaIOS, 2023-12-05:
I have to do something like this:
```python
cam = Andor.AndorSDK2Camera(idx=1, temperature = -30, fan_mode = "on")
print(cam.get_full_info())
time.sleep(10.)
print(cam.get_full_info()) 
```

or i get:
```
AndorSDK2LibError                         Traceback (most recent call last)
<ipython-input-28-a5801984caf1> in <module>
      5 import time
      6 
----> 7 cam = Andor.AndorSDK2Camera(idx=1)
      8 print(cam.get_full_info())
      9 time.sleep(10.)

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in __init__(self, idx, ini_path, temperature, fan_mode, amp_mode)
    117         self._start_fan_mode=fan_mode
    118         self._start_amp_mode=amp_mode
--> 119         self.open()
    120 
    121         self._device_var_ignore_error={"get":(AndorNotSupportedError,),"set":(AndorNotSupportedError,)}

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in open(self)
    252                     lib.Initialize(py3.as_builtin_bytes(self.ini_path))
    253                     self._opid=libctl.open().opid
--> 254                 self._setup_default_settings()
    255     def close(self):
    256         """Close connection to the camera"""

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in _setup_default_settings(self)
    180         try:
    181             self._strict_option_check=False
--> 182             self._initial_setup_temperature()
    183             self.init_amp_mode(mode=self._start_amp_mode)
    184             self.set_fan_mode(self._start_fan_mode)

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in _initial_setup_temperature(self)
    167             else:
    168                 self._start_temperature=0
--> 169             self.set_temperature(self._start_temperature,enable_cooler=True)
    170     def _initial_setup_ext_trigger(self):
    171         try:

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in wrapped(self, *args, **kwargs)
     73                     return self._cpar.get(getpar,defpar)
     74                 else:
---> 75                     res=func(self,*args,**kwargs)
     76                     if setpar is not None:
     77                         self._cpar[setpar]=res

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in set_temperature(self, temperature, enable_cooler)
    382         lib.SetTemperature(temperature)
    383         if enable_cooler:
--> 384             self.set_cooler(True)
    385         return temperature
    386     @_camfunc(getpar="temperature_setpoint",option=[("set",AC_SETFUNC.AC_SETFUNCTION_TEMPERATURE)])

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in wrapped(self, *args, **kwargs)
     73                     return self._cpar.get(getpar,defpar)
     74                 else:
---> 75                     res=func(self,*args,**kwargs)
     76                     if setpar is not None:
     77                         self._cpar[setpar]=res

~\anaconda3\lib\site-packages\pylablib\devices\Andor\AndorSDK2.py in set_cooler(self, on)
    349         else:
    350             lib.CoolerOFF()
[... 20 lines cut ...]
~\anaconda3\lib\site-packages\pylablib\core\utils\ctypes_wrap.py in wrapped_func(*vargs, **kwargs)
    274                     return a
    275             call_args=[_to_call_arg(n,t,a) for (n,t,a) in zip(argnames,prep_argtypes,func_args)]
--> 276             retval=func(*call_args)
    277             res=self._convert_results(rvals or [None],dict(zip(argnames,func_args)),retval,kwargs,rconv)
    278             if (not self.tuple_single_retval) and len(res)==0:

~\anaconda3\lib\site-packages\pylablib\devices\Andor\atmcd32d_lib.py in errchecker(result, func, arguments)
     39     def errchecker(result, func, arguments):  # pylint: disable=unused-argument
     40         if result not in passing:
---> 41             raise AndorSDK2LibError(func.__name__,result)
     42         return result
     43     return errchecker

AndorSDK2LibError: function 'IsCoolerOn' raised error 20081(DRV_I2CDEVNOTFOUND)
```
when i run the above, I get identical outputs:
```
{'image_indexing': 'rcb', 'frame_format': 'list', 'frame_info_format': 'namedtuple', 'frame_info_period': 1, 'roi': (0, 1024, 0, 127, 1, 1), 'temperature': None, 'channel': 0, 'oamp': 0, 'hsspeed': 0, 'preamp': 0, 'vsspeed': 0, 'shutter': ('closed', 0, 0, 0), 'fan_mode': None, 'trigger_mode': 'int', 'acq_parameters/accum': (1, 0), 'acq_parameters/kinetic': (1, 0.0, 1, 0, 0), 'acq_parameters/fast_kinetic': (1, 0.0), 'acq_parameters/cont': 0, 'acq_mode': 'cont', 'frame_transfer': None, 'read_parameters/single_track': (0, 1), 'read_parameters/multi_track': (1, 1, 0), 'read_parameters/random_track': [(0, 1)], 'read_parameters/image': (0, 1024, 0, 127, 1, 1), 'read_mode': 'image', 'exposure': 0.0010000000474974513, 'frame_period': 0.14000000059604645, 'buffer_size': 0, 'acquired_frames': 0, 'acquisition_in_progress': False, 'frames_status': TFramesStatus(acquired=0, unread=0, skipped=0, buffer_size=0), 'data_dimensions': (127, 1024), 'frame_timings': TAcqTimings(exposure=0.0010000000474974513, frame_period=0.14000000059604645), 'roi_limits': (TAxisROILimit(min=2, max=1024, pstep=1, sstep=1, maxbin=32), TAxisROILimit(min=2, max=127, pstep=1, sstep=1, maxbin=32)), 'temperature_monitor': 28.0, 'temperature_status': 'off', 'amp_mode': TAmpModeFull(channel=0, channel_bitdepth=16, oamp=0, oamp_kind='Conventional', hsspeed=0, hsspeed_MHz=1.0, preamp=0, preamp_gain=1. [... cut ...] Timings(exposure=0.0010000000474974513, accum_cycle_time=0.14000000059604645, kinetic_cycle_time=0.14000000059604645), 'readout_time': 9.999999747378752e-06, 'cls': 'AndorSDK2Camera', 'conn': (1, ''), 'detector_size': (1024, 127), 'frame_info_fields': ['frame_index'], 'device_info': TDeviceInfo(controller_model='CCI-010', head_model='DV401', serial_number=0), 'pixel_size': (2.6e-05, 2.6e-05)}
```

And i get the same timeout error as before.

Unfortunately, I do not have access to Andor software.

AlexShkarin, 2023-12-08:
OK, I see. Can you now try
```python
import time

cam = Andor.AndorSDK2Camera(idx=1,temperature = -30,fan_mode = "on")
print(cam.get_full_info())
cam.start_acquisition()
print(cam.get_full_info())
time.sleep(10.)
print(cam.get_full_info())
```
to see if any settings change when the acquisition is started?