AlexShkarin/pyLabLib #67: Live acquisition

khoroshyy, 2023-11-26:
Hi, I have a question.
Could you point me to a code where a live acquisition is handled?
I tried to get a Live image using Thorlabs cam using a while loop and snap (collecting it into a list for now), but I need to add a delay between consecutive acquisitions. Otherwise, the camera gets a timeout error sometimes.
Is there a better way to avoid it?
Is there a way to run the camera continuously and just get a last frame from time to time?
Thanks.
Petro.

AlexShkarin, 2023-11-28:
Hi Petro,

There's one basic example in the [documentation](https://pylablib.readthedocs.io/en/stable/devices/cameras_basics.html#acquisition-loop):
```python
# nframes=100 relates to the size of the frame buffer; the acquisition will continue indefinitely
cam.setup_acquisition(mode="sequence", nframes=100)  # could be combined with start_acquisition, or kept separate
cam.start_acquisition()
while True:  # acquisition loop
    cam.wait_for_frame()  # wait for the next available frame
    frame = cam.read_oldest_image()  # get the oldest image which hasn't been read yet
    # ... process and/or store frame ...
    if time_to_stop:
        break
cam.stop_acquisition()
```
Here the acquisition loop is continuously running on the camera and generating new frames which you can process or store in the processing loop. Keep in mind, that you'll need to deal with them fast enough, otherwise the acquisition buffer (which holds all acquired but not yet processed frames) might get overfilled.

If you're ok with losing some frames and just want to query the most recent one, then you can replace ``read_oldest_image`` with ``read_newest_image``. This ways the buffer might eventually get filled and the oldest acquired frames will be lost, but the newest frames should still always be available.

khoroshyy, 2023-11-29:
Thanks a lot.
Petro.

khoroshyy, 2023-12-07:
Hi,
I tried the following:

```python
# Initialize the Thorlabs camera
serial_number = "17162"  # Replace with your camera's serial number
camera = Thorlabs.ThorlabsTLCamera(serial=serial_number)

#very simple cycle
# nframes=100 relates to the size of the frame buffer; the acquisition will continue indefinitely
camera.setup_acquisition(nframes=100)  # could be combined with start_acquisition, or kept separate
camera.start_acquisition()
for i in range(10):  # acquisition loop
    camera.wait_for_frame()  # wait for the next available frame
    frame = camera.read_oldest_image()  # get the oldest image which hasn't been read yet
    # ... process frame ...
    print(i)
cam.stop_acquisition()
```
I had to remove the "mode=sequence", because TCL camera does not understand this keyword.
```python
0
1
2
3
4
5
6
7

---------------------------------------------------------------------------
ThorlabsTLCameraTimeoutError              Traceback (most recent call last)
Cell In[9], line 17
     15 camera.start_acquisition()
     16 for i in range(10):  # acquisition loop
---> 17     camera.wait_for_frame()  # wait for the next available frame
     18     frame = camera.read_oldest_image()  # get the oldest image which hasn't been read yet
     19     # ... process frame ...

File ~\anaconda\envs\BAM311\Lib\site-packages\pylablib\core\devio\interface.py:666, in use_parameters.<locals>.wrapper.<locals>.wrapped(*args, **kwargs)
    663 @functools.wraps(func)
    664 def wrapped(*args, **kwargs):
    665     all_args=parse_args(args,kwargs)
--> 666     res=func(**all_args)
    667     return parse_reply(res,args,kwargs)

File ~\anaconda\envs\BAM311\Lib\site-packages\pylablib\devices\interface\camera.py:262, in ICamera.wait_for_frame(self, since, nframes, timeout, error_on_stopped)
    260         to=fto if to is None else min(to,fto)
    261     if to is not None and to<=0:
--> 262         raise self.TimeoutError
    263     self._wait_for_next_frame(timeout=to,idx=acquired_frames)
    264 self._frame_counter.wait_done()

ThorlabsTLCameraTimeoutError: 
```
Unfortunately, I get errors.
Do you have any idea how to avoid it?

AlexShkarin, 2023-12-08:
That's strange. I've tried it on CS135MUN (which is, probably, the same as CS135MU as far as the readout electronics is concerned), and it works fine without any freezes, so it's probably not a fundamental problem with the code. Does the camera work with ThorCam? Which version of ThorCam do you have installed? Have you tried to reboot the PC and power cycle the camera?

khoroshyy, 2023-12-08:
Hi.
It is CS135MU, I just mistyped it.
I have two of them.
The problem comes with both.
It works with Thorcam without a problem (ver. 3.7.0.6).
What I may try is another computer.
Reboot did not help,  power cycle I did not try. I can access it only
remotely now.
Best.
Petro.

khoroshyy, 2023-12-15:
A bit of an update.
Power cycling did not help.
I have two cameras connected,
The timeout error disappeared from the code above when I disconnected one of the cameras. 
The rest I will try next year.

AlexShkarin, 2023-12-16:
Thanks for the update! I guess, it could be related to having two cameras... In principle, it should not be a problem, and I've seen the code work with two Thorlabs scientific cameras simultaneously. However, those were different models, which might make a difference.

khoroshyy, 2023-12-16:
Hi.
I hope the problem is with PC.
I use now a small brick PC, not very powerful and not many outputs.
Hopefully if I use another PC it will be fine.
If it is because two cameras, then I am I trouble, as I need to use both
cameras simultaneously :(
There is no problem when I use Thorcam software.
Best.
Petro.