AlexShkarin/pyLabLib #69: Failes on Linux platform with Ando SDK2 installed.

drmcnelson, 2023-12-12:
Fresh install on a fedora 38 platform with the Andor Linux SDK2 installed and verified.

Run the following test code from the pylablib installation page

`from pylablib.devices import Andor  # import Andor devices module
cam = Andor.AndorSDK2Camera()  # connect to Andor SDK2 camera (e.g., iXon)
cam.set_exposure(0.1)  # set exposure to 100ms
frame = cam.snap()  # grab a single frame
cam.close()  # close the connection
`
Produces the following error:

Traceback (most recent call last):
  File "/home/datacollection/TeensyDataAcquistion/pylablib_test/./andortest.py", line 4, in <module>
    cam = Andor.AndorSDK2Camera()  # connect to Andor SDK2 camera (e.g., iXon)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/datacollection/.local/lib/python3.11/site-packages/pylablib/devices/Andor/AndorSDK2.py", line 119, in __init__
    self.open()
  File "/home/datacollection/.local/lib/python3.11/site-packages/pylablib/devices/Andor/AndorSDK2.py", line 245, in open
    ncams=get_cameras_number()
          ^^^^^^^^^^^^^^^^^^^^
  File "/home/datacollection/.local/lib/python3.11/site-packages/pylablib/devices/Andor/AndorSDK2.py", line 39, in get_cameras_number
    libctl.preinit()
  File "/home/datacollection/.local/lib/python3.11/site-packages/pylablib/devices/utils/load_lib.py", line 240, in preinit
    self._do_preinit()
  File "/home/datacollection/.local/lib/python3.11/site-packages/pylablib/devices/utils/load_lib.py", line 211, in _do_preinit
    self.lib.initlib()
  File "/home/datacollection/.local/lib/python3.11/site-packages/pylablib/devices/Andor/atmcd32d_lib.py", line 65, in initlib
    self.lib=load_lib.load_lib(lib_names,locations=("parameter/andor_sdk2",solis_path,sdk2_path,"global"),error_message=error_message,call_conv="stdcall")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/datacollection/.local/lib/python3.11/site-packages/pylablib/devices/utils/load_lib.py", line 123, in load_lib
    raise OSError("DLLs are not available on non-Windows platform")
OSError: DLLs are not available on non-Windows platform

drmcnelson, 2023-12-12:
Further information,  the Python library provided in the Andors SDK2, has no issue with loading the library from either platform, as exemplified in the following lines from their atmcd.py

```
    def _load_library(self, userPath):
        if sys.platform == "linux":
            return cdll.LoadLibrary("libandor.so")
        elif sys.platform == "win32":
            etc.

            return windll.LoadLibrary(path)

```

AlexShkarin, 2023-12-16:
Hello!

There's currently no support for library-based devices (which includues all cameras) on Linux, since the PCs that I have access to only have Windows installed.

In the best case scenario, the only difference between the two platforms is the path to the library file, so all you need to do is to supply the correct path (the snippet you've provided also seems to imply that). You can do it somthing like this:
```python
import pylablib as pll
pll.par["devices/dlls/andor_sdk2"] = "libandor.so"  # might also be a goot idea to provide a full absolute path just in case
from pylablib.devices import Andor
cam = Andor.AndorSDK2Camera()
```
You would also need to comment out [these two lines](https://github.com/AlexShkarin/pyLabLib/blob/c3521c8ad2fa6064b3f8a117576ed3053b7fa69f/pylablib/devices/utils/load_lib.py#L122C37-L122C37) performing the OS check in ``pylablib/devices/utils/load_lib.py``: :
```python
if platform.system()!="Windows":
    raise OSError("DLLs are not available on non-Windows platform")
```
However, even if it loads the library, I can not guarantee that the code will work, since there could potentiall be differences within the library code between the two platforms. If you decide to test it out, let me know if it works.