AlexShkarin/pyLabLib #71: Parameters not set correctly after initializing Thorlabs BSC203

YellowFromEgg, 2024-01-19:
Hello all,

I am trying to connect my computer to the Thorlabs BSC203 using the pylabib.devices package. If I try to do this without connecting to the stages via the Kinesis software, the parameters, e.g. speed and homing parameters, are not set correctly and the stages do not behave as intended. If I connect to the stages via the Kinesis software and then close this software, the stages work as intended.

After reading the documentation, I understand that it should be possible to connect to the stages without using Kinesis. Am I misinterpreting this or is there a bug in my software?

This is an excerpt of the program code:
```
from pylablib.devices import Thorlabs as tl

device_list = tl.list_kinesis_devices() # contains BSC203 with serial number
chanel_no = 2
with tl.KinesisMotor("70213114", scale="HDR50" , is_rack_system='auto') as stage: # errors when not using auto for rack system
        stage.blink(channel=chanel_no) 
```

AlexShkarin, 2024-01-21:
Hello!

Generally speaking, when these devices powered off and then back on, their settings are reset to some default state. It is, in priniciple, possible to change this state but, as far as I am aware, Kinesis software does not to it by default. What happens is that the Kinesis software itself stores settings for each device (probably, based on its serial number) and automatically applies them whenever they are opened in Kinesis. This means that between the powering on and the first connection in the Kinesis software the devices settings are the defaults stored on the devices themsleves, and that is what you observe when using it with pylablib (the same should happen if you want to use devices manually, as is possible with, e.g., KDC101 motor controller). In principle, there's a way to change these setting using "Persist Settings to the Device" checkbox in the Kinesis software; it doesn't always work in my experience, but it's worth a try.

YellowFromEgg, 2024-01-22:
Thank you for your detailed information. I will be happy to try again later.

ChihYuLee, 2024-04-18:
Hello!

First of all, thames for maintaining this beautiful package!
I am also trying to control BSC203 in our lab. 
I am wondering if there is any way to use python code and Kinesis or Apt software simultaneously as opposed to shut down the software before initializing Pylablib? 
What I am trying to do is to manually move the stage via the software, hit enter on my python-scripted GUI to read position and add those positions into my dictionary. However, it seems that it's not feasible when I have both windows opened.