AlexShkarin/pyLabLib #75: Problem when opening connection with zelux (thorlabs camera)

jupiterMJM, 2024-02-15:
I don't succeed in setting up a connection with a zelux camera (from thorlabs). 
For example, when i run the following code:
```python
from pylablib.devices import Thorlabs
import numpy as np

ad_serial = Thorlabs.list_cameras_tlcam()[0]
print(ad_serial, type(ad_serial))
cam1 = Thorlabs.TLCamera.ThorlabsTLCamera(ad_serial)
```

I get the following error:
```cmd
Traceback (most recent call last):
  File "c:\parcours_recherche\codes_python\projet2_extract_phase_from_mach_and_zender\prise_en_main.py", line 27, in <module>
    cam1 = Thorlabs.TLCamera.ThorlabsTLCamera(ad_serial)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\parcours_recherche\codes_python\projet2_extract_phase_from_mach_and_zender\.venv\Lib\site-packages\pylablib\devices\Thorlabs\TLCamera.py", line 74, in __init__
    self.open()
  File "c:\parcours_recherche\codes_python\projet2_extract_phase_from_mach_and_zender\.venv\Lib\site-packages\pylablib\devices\Thorlabs\TLCamera.py", line 110, in open
    self.set_color_format()
  File "c:\parcours_recherche\codes_python\projet2_extract_phase_from_mach_and_zender\.venv\Lib\site-packages\pylablib\core\devio\interface.py", line 666, in wrapped
    res=func(**all_args)
        ^^^^^^^^^^^^^^^^
  File "c:\parcours_recherche\codes_python\projet2_extract_phase_from_mach_and_zender\.venv\Lib\site-packages\pylablib\devices\Thorlabs\TLCamera.py", line 205, in set_color_format
    cinfo=self.get_color_info()
          ^^^^^^^^^^^^^^^^^^^^^
  File "c:\parcours_recherche\codes_python\projet2_extract_phase_from_mach_and_zender\.venv\Lib\site-packages\pylablib\core\devio\interface.py", line 667, in wrapped
    return parse_reply(res,args,kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\parcours_recherche\codes_python\projet2_extract_phase_from_mach_and_zender\.venv\Lib\site-packages\pylablib\core\devio\interface.py", line 654, in parse_reply
    if len(returns)!=len(res):
                     ^^^^^^^^
TypeError: object of type 'NoneType' has no len()
```

I don't understand why (due to the fact that i am following the documentation). Note that i can find the camera thanks to Thorlabs.list_cameras_tlcam(); and i can open it thanks to ThorCam.
Thanks for your help!

khoroshyy, 2024-02-18:
Not sure if it will solve the problem but instead of  (ad_serial) you can try (serial='36474')
Or whatever serial number you have.
I am out of lab and not sure what is output of list_cameras_tclcam, and cannot see output of your print statement. If it is a list you are feeling a list instead of string.
Best 
Petro

khoroshyy, 2024-02-18:
OK. I can see now you read the first item of the list.
You can try cam contol application which uses pylablib and see if it works.

AlexShkarin, 2024-02-18:
This is a bug in the most recent pylablib version (1.4.2). You can either try to downgrade to an earlier version, or use the solution described in #65.