AlexShkarin/pyLabLib #76: Connection problem with MLS203 thorlabs stage and BBD302 controller

MassiliaHamdani, 2024-02-16:
Hello, I can't establish a connection with the Thorlabs ML203 board (BBD302 controller).

When I run the following code :

AlexShkarin, 2024-02-18:
Hi!

It looks like the ID you use to connect (``104390055``) is different from the one reported by the ``list_kinesis_devices`` function (``103390054``).

nusratc2, 2024-03-15:
Hi Alex, I have been trying this command after seeing the controller is visible in kinesis software, but getting this error repeatedly, I can't connect the BBD 302 controller to the MLS stages too.

My code:
from pylablib.devices import Thorlabs
Thorlabs.list_kinesis_devices()
[('103387864', 'Kinesis Benchless Motor')]
stage = Thorlabs.KinesisMotor("103387864", is_rack_system=True)
stage = Thorlabs.KinesisMotor("103387864")

The error:

---------------------------------------------------------------------------
D2XXException                             Traceback (most recent call last)
File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:809, in FT232DeviceBackend.__init__(self, conn, timeout, term_write, term_read, open_retry_times, datatype, reraise_error)
    808 try:
--> 809     self.instr=self._open_instr(port,conn_dict)
    810     self.opened=True

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:821, in FT232DeviceBackend._open_instr(self, port, params)
    820 if "serial_number" in sig.arg_names: # pyft232 v0.11 signature change
--> 821     return ft232.Ft232(serial_number=port,**params)
    822 else:

File ~\AppData\Roaming\Python\Python311\site-packages\ft232\d2xx.py:127, in D2xx.__init__(self, port, serial_number, description, baudrate, bytesize, parity, stopbits, timeout, xonxoff, rtscts, writeTimeout)
    125     status = d2xx.FT_OpenEx(serial, FT_OPEN_BY_SERIAL_NUMBER,
    126                             c.byref(self.handle))
--> 127     if status != FT_OK: raise D2XXException(status)
    128 elif description:

D2XXException: Device Not Opened

The above exception was the direct cause of the following exception:

ThorlabsBackendError                      Traceback (most recent call last)
Cell In[16], line 4
      2 Thorlabs.list_kinesis_devices()
      3 [('103387864', 'Kinesis Benchless Motor')]
----> 4 stage = Thorlabs.KinesisMotor("103387864", is_rack_system=True)
      5 stage = Thorlabs.KinesisMotor("103387864")

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:1102, in KinesisMotor.__init__(self, conn, scale, default_channel, is_rack_system)
   1101 def __init__(self, conn, scale="step", default_channel=1, is_rack_system=False):
-> 1102     super().__init__(conn,default_channel=default_channel,is_rack_system=is_rack_system)
   1103     self.add_background_comm(0x0464) # move completed
   1104     self.add_background_comm(0x0466) # move stopped

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:277, in KinesisDevice.__init__(self, conn, timeout, default_channel, is_rack_system)
    276 def __init__(self, conn, timeout=3., default_channel=1, is_rack_system=False):
--> 277     super().__init__(conn,timeout=timeout,is_rack_system=is_rack_system,default_axis=default_channel)
    278     self._remove_device_variable("axes")
    279     self._add_info_variable("channel",self.get_all_channels)

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\interface\stage.py:34, in IMultiaxisStage.__init__(self, default_axis, *args, **kwargs)
     33 def __init__(self, *args, default_axis="all", **kwargs):
---> 34     super().__init__(*args,**kwargs)
     35     self._original_axis_parameter=None
     36     self._default_axis=default_axis

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:44, in BasicKinesisDevice.__init__(self, conn, timeout, is_rack_system)
     42 def __init__(self, conn, timeout=3., is_rack_system=False):
     43     defaults={"serial":{"baudrate":115200,"rtscts":True}, "ft232":{"baudrate":115200,"rtscts":True}}
---> 44     instr=comm_backend.new_backend(conn,backend=("auto","ft232"),term_write=b"",term_read=b"",timeout=timeout,
     45         defaults=defaults,reraise_error=ThorlabsBackendError)
     46     instr.setup_cooldown(write=0.003)
     47     try:

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:1776, in new_backend(conn, backend, defaults, **kwargs)
   1774 if defaults is not None and backend_name is not None and backend_name in defaults:
   1775     conn=backend.combine_conn(conn,defaults[backend_name])
-> 1776 return backend(conn,**kwargs)

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:816, in FT232DeviceBackend.__init__(self, conn, timeout, term_write, term_read, open_retry_times, datatype, reraise_error)
    814     self._conn_params=(port,conn_dict,timeout)
    815 except self.BackendError as e:
--> 816     raise self.Error(e) from e

ThorlabsBackendError: backend exception: D2XXException(3) ('Device Not Opened')

stage = Thorlabs.KinesisMotor("103387864", is_rack_system=True)

C:\Users\nusratc2\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:1230: UserWarning: can't recognize motor model BBD302; setting all scales to internal units
  warnings.warn("can't recognize motor model {}; setting all scales to internal units".format(self._model)) 

Can you please help me with it. I am stuck for days.

nusratc2, 2024-03-15:
Hi Alex, I have been trying this command after seeing the controller is visible in kinesis software, but getting this error repeatedly, I can't connect the BBD 302 controller to the MLS stages too. Everytime I run this code, the controller is not visible by the kinesis software anymore, I have to restart the device again, is there any way to establish this connection and not lose it due after every run?

My code:
from pylablib.devices import Thorlabs
Thorlabs.list_kinesis_devices()
[('103387864', 'Kinesis Benchless Motor')]
stage = Thorlabs.KinesisMotor("103387864", is_rack_system=True)
stage = Thorlabs.KinesisMotor("103387864")

The error:

D2XXException Traceback (most recent call last)
File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:809, in FT232DeviceBackend.init(self, conn, timeout, term_write, term_read, open_retry_times, datatype, reraise_error)
808 try:
--> 809 self.instr=self._open_instr(port,conn_dict)
810 self.opened=True

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:821, in FT232DeviceBackend._open_instr(self, port, params)
820 if "serial_number" in sig.arg_names: # pyft232 v0.11 signature change
--> 821 return ft232.Ft232(serial_number=port,**params)
822 else:

File ~\AppData\Roaming\Python\Python311\site-packages\ft232\d2xx.py:127, in D2xx.init(self, port, serial_number, description, baudrate, bytesize, parity, stopbits, timeout, xonxoff, rtscts, writeTimeout)
125 status = d2xx.FT_OpenEx(serial, FT_OPEN_BY_SERIAL_NUMBER,
126 c.byref(self.handle))
--> 127 if status != FT_OK: raise D2XXException(status)
128 elif description:

D2XXException: Device Not Opened

The above exception was the direct cause of the following exception:

ThorlabsBackendError Traceback (most recent call last)
Cell In[16], line 4
2 Thorlabs.list_kinesis_devices()
3 [('103387864', 'Kinesis Benchless Motor')]
----> 4 stage = Thorlabs.KinesisMotor("103387864", is_rack_system=True)
5 stage = Thorlabs.KinesisMotor("103387864")

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:1102, in KinesisMotor.init(self, conn, scale, default_channel, is_rack_system)
1101 def init(self, conn, scale="step", default_channel=1, is_rack_system=False):
-> 1102 super().init(conn,default_channel=default_channel,is_rack_system=is_rack_system)
1103 self.add_background_comm(0x0464) # move completed
1104 self.add_background_comm(0x0466) # move stopped

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:277, in KinesisDevice.init(self, conn, timeout, default_channel, is_rack_system)
276 def init(self, conn, timeout=3., default_channel=1, is_rack_system=False):
--> 277 super().init(conn,timeout=timeout,is_rack_system=is_rack_system,default_axis=default_channel)
278 self._remove_device_variable("axes")
279 self._add_info_variable("channel",self.get_all_channels)

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\interface\stage.py:34, in IMultiaxisStage.init(self, default_axis, *args, **kwargs)
33 def init(self, *args, default_axis="all", **kwargs):
---> 34 super().init(*args,**kwargs)
35 self._original_axis_parameter=None
36 self._default_axis=default_axis

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:44, in BasicKinesisDevice.init(self, conn, timeout, is_rack_system)
42 def init(self, conn, timeout=3., is_rack_system=False):
43 defaults={"serial":{"baudrate":115200,"rtscts":True}, "ft232":{"baudrate":115200,"rtscts":True}}
---> 44 instr=comm_backend.new_backend(conn,backend=("auto","ft232"),term_write=b"",term_read=b"",timeout=timeout,
45 defaults=defaults,reraise_error=ThorlabsBackendError)
46 instr.setup_cooldown(write=0.003)
47 try:

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:1776, in new_backend(conn, backend, defaults, **kwargs)
1774 if defaults is not None and backend_name is not None and backend_name in defaults:
1775 conn=backend.combine_conn(conn,defaults[backend_name])
-> 1776 return backend(conn,**kwargs)

File ~\AppData\Roaming\Python\Python311\site-packages\pylablib\core\devio\comm_backend.py:816, in FT232DeviceBackend.init(self, conn, timeout, term_write, term_read, open_retry_times, datatype, reraise_error)
814 self._conn_params=(port,conn_dict,timeout)
815 except self.BackendError as e:
--> 816 raise self.Error(e) from e

ThorlabsBackendError: backend exception: D2XXException(3) ('Device Not Opened')

stage = Thorlabs.KinesisMotor("103387864", is_rack_system=True)

C:\Users\nusratc2\AppData\Roaming\Python\Python311\site-packages\pylablib\devices\Thorlabs\kinesis.py:1230: UserWarning: can't recognize motor model BBD302; setting all scales to internal units
warnings.warn("can't recognize motor model {}; setting all scales to internal units".format(self._model))

Can you please help me with it. I am stuck for days. Please help me.

AlexShkarin, 2024-03-18:
Hello!

Every time after opening the device, you need to close it again:
```python
stage = Thorlabs.KinesisMotor("103387864", is_rack_system=True)
# ... do some work ...
stage.close()
```
This signals the OS that other software (e.g., Kinesis) can connect to it.
In order to ensure that the device gets closed even when an error occurs, you can use the context manager approach:
```python
with Thorlabs.KinesisMotor("103387864", is_rack_system=True) as stage:
    # ... do some work ...
# the stage closes automatically when the execution is done
```
Also, keep in mind that only one connection can be opened at a time, which means that the code you have right now definitely won't work: it tries to connect to the same device 2 times in a row, so the second line will raise an error.

nusratc2, 2024-03-21:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[33], line 3
      1 import pylablib.devices.Thorlabs.kinesis
      2 from pylablib.devices import Thorlabs
----> 3 Thorlabs.list_kinesis_devices()

File ~\.conda\envs\API\Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py:26, in list_kinesis_devices(filter_ids)
     18 def list_kinesis_devices(filter_ids=True):
     19     """
     20     List all Thorlabs APT/Kinesis devices connected to this PC.
     21 
   (...)
     24         Otherwise, show all devices (some of them might not be Thorlabs-related).
     25     """
---> 26     return KinesisDevice.list_devices(filter_ids=filter_ids)

File ~\.conda\envs\API\Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py:101, in BasicKinesisDevice.list_devices(filter_ids)
     99 def _is_thorlabs_id(did):
    100     return re.match(r"^\d{8}$",did[0]) is not None
--> 101 ids=comm_backend.FT232DeviceBackend.list_resources(desc=True)
    102 if filter_ids:
    103     ids=[did for did in ids if _is_thorlabs_id(did)]

AttributeError: module 'pylablib.core.devio.comm_backend' has no attribute 'FT232DeviceBackend'

Why I am getting this error when I try to run it? How to fix this one? Can you please help me?