AlexShkarin/pyLabLib #79: Arcus Performax Stage AttributeError: NoneType object has no attribute 'argtypes'

iandc808, 2024-02-28:
I've been trying to control an Arcus Technology DMX-J-SA-17 using but whenever I attempt to run my script I'm met with an error declaring: AttributeError: NoneType object has no attribute 'argtypes'. Here is what I have so far:

import pylablib

pylablib.par['devices/dlls/arcus_performax']='C:/Users/idcornw/Desktop/Gyro Code/bin/Py2.7.2_win32/DLLs/PerformaxCom.dll'
pylablib.par['devices/dlls/arcus_performax']='C:/Users/idcornw/Desktop/Gyro Code/bin/Py2.7.2_win32/DLLs/SiUSBXp.dll'

from pylablib.devices import Arcus

Arcus.list_usb_performax_devices()

Any help would be appreciated.

Thank you,
Ian

iandc808, 2024-02-28:
Actually, sorry, let me clarify. I'm using a Newmark Systems NSC-A1, but it supposedly works on the same drivers and commands and such.

AlexShkarin, 2024-03-02:
I think the problem is in the second line:
```python
pylablib.par['devices/dlls/arcus_performax']='C:/Users/idcornw/Desktop/Gyro Code/bin/Py2.7.2_win32/DLLs/SiUSBXp.dll'
```
It overwrites the path to the dll that was set before, so now it points to the wrong library. Since this library does not have the necessary functions, it ends up raising an error.
I would suggest getting rid of this line, i.e., leave only
```python
pylablib.par['devices/dlls/arcus_performax']='C:/Users/idcornw/Desktop/Gyro Code/bin/Py2.7.2_win32/DLLs/PerformaxCom.dll'
```

Let me know if that helps!

Sincerely,

Alexey.

iandc808, 2024-04-11:
Very sorry for the late reply, but that totally fixed it. Thank you for the help and for making the whole thing, it's great.