AlexShkarin/pyLabLib #86: BPC301 - connected, but no communication

JOBowser, 2024-05-01:
I am trying to connect to a BPC301 APT Piezo Controller using this library. I successfully initialized the motor and called get_device_info() with:
```
bpc301 =  Thorlabs.KinesisMotor("41878356", is_rack_system= 'auto')
print(bpc301._is_rack_system)
print(bpc301.get_device_info())
```
giving me the printout: 
```
auto
TDeviceInfo(serial_no=41878356, model_no='BPC301', fw_ver='3.2.2', hw_type=18, hw_ver=3, mod_state=0, nchannels=1, notes='APT Piezo Controller')
```
However, when I try to call methods like get_position(), get_status(), or move_by(), I get the following error:

```
Traceback (most recent call last):
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\playground.py", line 122, in <module>
    pylablib_motors()
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\playground.py", line 94, in pylablib_motors
    print(bpc301.get_status())
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\core\utils\general.py", line 1107, in wrapped
    res=func(**all_args)
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 356, in _get_status
    status_n=self._get_status_n(channel=channel)
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\core\utils\general.py", line 1107, in wrapped
    res=func(**all_args)
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 330, in _get_status_n
    data=self.query(self._status_comm,self._make_channel(channel),dest=("channel",channel)).data
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 185, in query
    return self.recv_comm(expected_id=replyID)
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 136, in recv_comm
    b=self.instr.read(1)
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\core\devio\comm_backend.py", line 45, in wrapped
    return func(self,*args,**kwargs)
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\core\devio\comm_backend.py", line 34, in wrapped
    return func(self,*args,**kwargs)
  File "c:\Users\aaa\Documents\GitHub\Visual_Inspection\.venv\lib\site-packages\pylablib\core\devio\comm_backend.py", line 925, in read
    raise self.Error("read returned less data than expected")
pylablib.devices.Thorlabs.base.ThorlabsBackendError: backend exception: 'read returned less data than expected' ('read returned less data than expected')
```
I have tried calling KinesisPiezoMotor instead, but nothing changed. This is a single channel piezo controller.

AlexShkarin, 2024-05-04:
It looks like BPC301 is a continuous piezo driver, which is not supporent in the currently released version of pylablib (``KinesisPiezoMotor`` are piezo-intertial slip-stick motor drivers like KIM001, which operate differently).

However, there's some support for relatively similar KPZ101 drivers in the currently developed version. You can access it by replacing the ``lib\site-packages\pylablib\devices\Thorlabs\kinesis.py`` file with [this version](https://github.com/AlexShkarin/pyLabLib/blob/dev/pylablib/devices/Thorlabs/kinesis.py) from GitHub and using ``KinesisPiezoController`` class:
```python
from pylablib.devices.Thorlabs import kinesis
dev=Thorlabs.KinesisPiezoController("41878356",is_rack_system="auto")
print(dev.get_output_voltage())
```
Let me know if that works.

JOBowser, 2024-05-07:
Thank you for the response. Unfortunately, this did not work. kinesis.KinesisPiezoController() does not take a is_rack_system argument. Even without that argument, I am still getting the same error as before: 

> pylablib.devices.Thorlabs.base.ThorlabsBackendError: backend exception: 'read returned less data than expected' ('read returned less data than expected')

I've managed to use the Thorlabs Github example, plus my own testing to get something working for this motor controller. If you need me to test any more code, I'd be happy to help.