AlexShkarin/pyLabLib #87: Issue with jogging channel 3 of Kinesis Piezomoter with KIM101 4 channel controller

knuxls, 2024-05-04:
Hello - 

First, thank you for writing this interface software for so many types of lab equipment, it is very helpful. 

I think I've found a bug: when I try to jog any channel 3 on a Thorlab's Kinesis KIM101 controller, I get an error. However other channels and all the channel the move_by commands work with the same controllers. Is this a known issue? 

The offending line in kinesis.py:
channel={0x01:1,0x02:2,0x04:3,0x08:4}[channel]

no channel 3 to key off of? 
==============

Example code:

from pylablib.devices import Thorlabs
import time
import traceback

kcube = Thorlabs.KinesisPiezoMotor(sn)

chan = 3
try:
    print(f'rel move + k1: ch:{chan}')
    kcube.move_by(200, channel=chan)
    time.sleep(2)
    print(f'rel move - k1: ch:{chan}')
    kcube.move_by(-200, channel=chan)
    time.sleep(2)
    print('')
except Exception:
    traceback.print_exc()

try:
    print(f'jog + k1: ch:{chan}')
    kcube.jog('+', channel=chan, kind='continuous')
    time.sleep(2)
    print(f'jog - k1: ch:{chan}')
    kcube.jog('-', channel=chan, kind='continuous')
    time.sleep(2)
    print(f'stop k1: ch:{chan}')
    kcube.stop(chan)
    print('')
except Exception:
    traceback.print_exc()

kcube1.close()

==============

output

==============

rel move + k1: ch:3
rel move - k1: ch:3

jog + k1: ch:3
Traceback (most recent call last):
  File "Z:\python_scripts\dev\testch3.py", line 28, in <module>
    kcube.jog('+', channel=chan, kind='continuous')
  File "C:\ProgramData\anaconda3\Lib\site-packages\pylablib\core\utils\general.py", line 1107, in wrapped
    res=func(**all_args)
        ^^^^^^^^^^^^^^^^
  File "C:\ProgramData\anaconda3\Lib\site-packages\pylablib\core\devio\interface.py", line 666, in wrapped
    res=func(**all_args)
        ^^^^^^^^^^^^^^^^
  File "C:\ProgramData\anaconda3\Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 970, in _pzmot_jog
    self._pzmot_autoenable(channel,auto_enable)
  File "C:\ProgramData\anaconda3\Lib\site-packages\pylablib\devices\Thorlabs\kinesis.py", line 941, in _pzmot_autoenable
    channel={0x01:1,0x02:2,0x04:3,0x08:4}[channel]
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
KeyError: 3

AlexShkarin, 2024-05-04:
You're right, there's a bug in the current implementation of the ``jog`` method. I'll fix it in the next release, but for now you can get around it by setting ``auto_enable=False`` in the call and enabling the axis explicitly before. In other words, instead of
```python
kcube.jog('+', channel=chan, kind='continuous')
```
you would call
```python
kcube.enable_channels(channel=chan)
kcube.jog('+', channel=chan, kind='continuous', auto_enable=False)
```
Let me know if that works.

knuxls, 2024-05-07:
The code doesn’t throw an error with this change. But unfortunately, neither does it jog the motor. It doesn’t jog any channels actually… So, trying to manually enable and jog works for no channels. Auto enable works for most channels, except channel 3. kcube.get_status(channel=3) returns an empty string after kcube.enable_channels(channel=3). Maybe it isn’t being enabled?Cheers