AlexShkarin/pyLabLib #93: Unable to use Photometrics and PrincetonInstruments in the same time

llooiicc, 2024-11-06:
Hello,

I try to use 2 cameras a PrincetonInstruments and a Photometrics. But I it seems it is not allowed to load this two dll in the same time.

I have this short code :
```
from pylablib.devices import Photometrics
from pylablib.devices import PrincetonInstruments

if __name__ == "__main__":
    cams = PrincetonInstruments.list_cameras()
    cam1 = PrincetonInstruments.PicamCamera(cams[0].serial_number)

    cams = Photometrics.list_cameras()
    cam2 = Photometrics.PvcamCamera(cams[0])
```

And the errors returned :

```
Traceback (most recent call last):
  File "d:\test_pylablib\main.py", line 25, in <module>
    cams = Photometrics.list_cameras()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\test_pylablib\.venv\Lib\site-packages\pylablib\devices\Photometrics\pvcam.py", line 39, in list_cameras
    with libctl.temp_open():
         ^^^^^^^^^^^^^^^^^^
  File "C:\Users\emac\AppData\Local\Programs\Python\Python312\Lib\contextlib.py", line 137, in __enter__
    return next(self.gen)
           ^^^^^^^^^^^^^^
  File "D:\test_pylablib\.venv\Lib\site-packages\pylablib\devices\utils\load_lib.py", line 290, in temp_open
    init_result,open_result,opid=self.open()
                                 ^^^^^^^^^^^
  File "D:\test_pylablib\.venv\Lib\site-packages\pylablib\devices\utils\load_lib.py", line 255, in open
    init_result=self._do_init() # pylint: disable=assignment-from-no-return
                ^^^^^^^^^^^^^^^
  File "D:\test_pylablib\.venv\Lib\site-packages\pylablib\devices\Photometrics\pvcam.py", line 31, in _do_init
    lib.pl_pvcam_init()
  File "<string>", line 1, in <lambda>
  File "D:\test_pylablib\.venv\Lib\site-packages\pylablib\core\utils\ctypes_wrap.py", line 276, in wrapped_func
    retval=func(*call_args)
           ^^^^^^^^^^^^^^^^
  File "D:\test_pylablib\.venv\Lib\site-packages\pylablib\devices\Photometrics\pvcam_lib.py", line 41, in errchecker
    raise PvcamLibError(func.__name__,lib=lib)
pylablib.devices.Photometrics.pvcam_lib.PvcamLibError: function 'pl_pvcam_init' raised error 158: The initialization has been already done (PL_ERR_LIBRARY_ALREADY_INITIALIZED)
```

AlexShkarin, 2025-01-08:
Hi, sorry for the late reply!

Unfortunately, there's not much I can suggest here. I've seen this behavior once before, but I could not figure our any solution. It really seems to be some kind of low-level conflict between the camera drivers. The only way I can see addressing this is to run two separate scripts or, perhaps, using multiprocessing (though I'm not sure about this one) so that one process loads only one kind of camera at a time.

Malphas64, 2025-01-09:
I've had some success using a Princeton ProEM at the same time as a Photometrics Kinetix by installing PICAM drivers first and then PVCAM drivers on top of it, hope it helps!