AlexShkarin/pyLabLib #99: Advice: implementing support for AndorSDK3+ on Linux

casparvitch, 2025-02-06:
Hiya, thanks for the nice package. Andor cameras working nicely on windows.

We're attempting to move to linux, and so I'm digging into pylablib to see if it's simple to swap to .so libs. Also note #69. 

I specify the lib location with:
```python
pll.par["devices/only_windows_dlls"] = False
pll.par["devices/dlls/andor_sdk3"] = "/usr/local/lib/libatcore.so"
```

So far the roadblocks I've come up against:
- Even with the `only_windows_dlls` lib param set to `False`, you're blocked at [`load_lib` here](https://github.com/AlexShkarin/pyLabLib/blob/000e5ddd59e22f7f9fe6a9ce318c0721a176b601/pylablib/devices/utils/load_lib.py#L123) (easy change, remove the 'not')
- Am I correct in assuming that the `load_lib` function should be called with the `cdecl` `call_conv`, e.g. [here](https://github.com/AlexShkarin/pyLabLib/blob/000e5ddd59e22f7f9fe6a9ce318c0721a176b601/pylablib/devices/Andor/atcore_lib.py#L59)?
- is `ctypes.WINFUNCTYPE` → `ctypes.CFUNCTYPE` correct? e.g. [here](https://github.com/AlexShkarin/pyLabLib/blob/000e5ddd59e22f7f9fe6a9ce318c0721a176b601/pylablib/devices/Andor/atcore_lib.py#L135)

I will continue attempting to sidestep these issues, but if you have any guidance that would be appreciated 🙏

I am happy to lodge a PR eventually (or not) with a clean solution for a couple of camera types, but if you would like a more general solution (across other devices), I am happy to discuss that too.

AlexShkarin, 2025-02-07:
Thanks for bringing these issues up! I'll fix them by the next release. What you are suggesting (replacing ``windll`` with ``cdecl`` and ``WINFUNCTYPE`` with ``CFUNCTYPE`` seem correct to me. And the ``only_windows_dlls`` line is, indeed, a typo on my part.

Let me know how this works out, and if you uncover any other issues!

casparvitch, 2025-02-10:
Works well for all of the Andor cameras I have available so far 🚀 
I'll let you know if there's any other issues, especially when I go to other manufacturers. Cheers

closed 2025-02-10 by casparvitch